FreeCAD/FreeCAD #20356: Incorrect Collada pruning when exporting BIM model

Ivan2309, 2025-03-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the attached file, I exported as Collada format (.dae). FreeCAD reports several instances of pruning the "roof" and, indeed, the roof is missing when I import the file in Blender or re-import to FreeCAD.

The file comes from the FCB Lounge youtube channel.

User roy_043 appears to have identified the root cause and wrote:
"The problem is caused by the roof being used to subtract from the walls. As a workaround you can create a [Part_SimpleCopy](https://wiki.freecad.org/Part_SimpleCopy) from the roof and include that in your export."

This suggestion solved the immediate problem but, I think, could get very tedious with a more complicated model.

Kind regards,

Ivan.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.47
  * A2plus 0.4.67
  * fasteners 0.5.29
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?style=1&t=95709
https://youtu.be/WZHyUBfdgJA?t=2506

[garage_3D_end (Copy).zip](https://github.com/user-attachments/files/19398232/garage_3D_end.Copy.zip)

I only renamed the file, rather than actually zipping it.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Ivan2309, 2025-04-08:
Is this related to https://github.com/FreeCAD/FreeCAD/issues/18107?

My report and 18107 both note that the roof is missing after export even though we are using two different formats.

ErtugrulEge, 2025-05-04:
This is not happening just about collada format, also if you choose *.obj format, roofs are pruning

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
```

furgo16, 2025-05-17:
Also related forum thread: https://forum.freecad.org/viewtopic.php?p=827040