FreeCAD/FreeCAD #20367: Application positioning problem

Asger-P, 2025-03-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using windows large font (125%) and two screens. If the FreeCAD window was positioned on the center of the right (second) screen, filling 95% of the screen, when it was closed. FreeCAD will be positioned wrong when opened again see pic one and two

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Denmark (en_DK)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.37
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Maybe related https://github.com/FreeCAD/FreeCAD/issues/11142

Asger-P, 2025-03-26:
> Maybe related [#11142](https://github.com/FreeCAD/FreeCAD/issues/11142)

I don't think so, that issue sound more like not referencing the right screen when positioning the menu.
The problem I show is related to scaling the app position at start-up.