FreeCAD/FreeCAD #20371: Setting properties with Hidden does not persist relaunching the Freecad/file.

leoheck, 2025-03-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Setting a property to `Hidden`, does not persist after relaunching the file.
This works with the `ReadOnly`.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Please add an example file and steps how to reproduce.
What object types are affected and how do you set the property?

leoheck, 2025-10-03:
@maxwxyz this is the property I could not make invisible, since when the document is opened again, it appears visible again.

https://github.com/leoheck/FreeCAD_Assembly4.1/blame/6776c60dafe2a22feed24c270dcfef532b533bf4/freecad/Asm4p1/make_array_cmd.py#L215

FEA-eng, 2026-03-04:
@leoheck So it's specific to Assembly4 and you haven't noticed this behavior with any core properties?

FEA-eng, 2026-09-03:
@leoheck Please confirm if it's still relevant and specific to Assembly4.

closed 2026-09-03 by maxwxyz

GeeBee2, 2026-09-03:
The same issue came up today on the [forum](https://forum.freecad.org/viewtopic.php?p=904547). It is not specific to Assembly4. 
[tn908.FCStd.zip](https://github.com/user-attachments/files/31795036/tn908.FCStd.zip)
The visibilty of 'hilfsgeometrie' needs to be changed to hidden and saved. When reopened it is visible again.

FEA-eng, 2026-09-07:
> The same issue came up today on the [forum](https://forum.freecad.org/viewtopic.php?p=904547).

Is it really the same thing, though? The issue here is about the hidden status of properties (in the Property View), while the forum thread mentions hidden objects/geometries.

GeeBee2, 2026-09-08:
> The issue here is about the hidden status of properties (in the Property View), while the forum thread mentions hidden objects/geometries.

Yes you are right. I misunderstood the issue.