FreeCAD/FreeCAD #20375: BIM_Classification - 'Classification' folder not created in default FreeCAD installation

semhustej, 2025-03-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**
`Classification` folder is not created in default FreeCAD installation on Windows. When user wants to use classification functionality in FreeCAD, he/she has to create the `Classification` folder manually to be able to copy classification file into it. 

**Proposed solution**
The folder `Classification` should be created in `C:\Users\'username'\AppData\Roaming\FreeCAD\BIM` during FreeCAD installation

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-24:
No strong opinion on this... I am a bit reluctant to create folders on people's computers without maybe ever needing them, but it's also no big harm...

semhustej, 2025-03-24:
> No strong opinion on this... I am a bit reluctant to create folders on people's computers without maybe ever needing them, but it's also no big harm...

It is not a big deal, but it would be nice.

I don't think users would care much about another folder in FreeCAD's roaming folder structure.

My user experience using classification for the first time: I was a bit confused that I have to manually create the folder. At first I thought I am in a wrong location and was not sure I am doing the right steps.

If the folder is not created automatically, it would be nice to mentioned on the wiki, that the folder has to be created manually.

yorikvanhavre, 2025-05-13:
I edited https://wiki.freecad.org/BIM_Classification for the time being.

Still not too sure if we should do this or not, so leaving the issue open for now..

yorikvanhavre, 2025-05-15:
Oh sorry... I have overlooked a bit my forum inbox 😅 Will do!

Okay, closing this one then!