FreeCAD/FreeCAD #20377: Assembly: Solver goes unstable and blows up assembled parts

DavidLister, 2025-03-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was tinkering with assemblies and put my assembly into an unstable state. When I go to solve it now, the solver no longer converges and the parts blow up.

In the attached file, open the assembly in assembly workbench and then solve it using `z` or the UI button. Each button press will not converge and leave the assembly in a more broken state than before. After several iterations, it stops making progress and reports this error message:

```
12:53:15  MbD: Convergence = 1373168948.6613345
12:53:15  MbD: Convergence = 835241872.90007353
12:53:15  MbD: Convergence = 13384569.624690192
12:53:15  MbD: Convergence = 3719356.1130926087
12:53:15  MbD: Convergence = 22895061.188299287
12:53:15  MbD: No convergence after 101 iterations.
12:53:15  MbD: A geometrically incompatible system has been specified.
12:53:15  MbD: Or the system parts are distributed too far apart from the assembled positions.
12:53:15  0.77174 <Assembly> AssemblyObject.cpp(176): Solve failed: iterNo > iterMax
```

Removing the Cylindrical constraint and holding `z` give interesting behavior as well, rotating the links around the fixed one for a long time. Even deleting all but the first Revolute constraint doesn't give the solver a starting condition that will converge.

If this was an important project, I'd be stuck and have to redo the assembly which for complex ones would take a long time. I think it's important that the solver be robust to this type of pertubation natively, or give the use options to edit the solver's parameters similar to what's available within the sketch workbench.

AS for how this state was created. I was trying out Assembly3 and Assembly4 workbenches in the same file, then deleted the assemblies associated with those in the model tree. Upon deleting them, the Assembly workbench's assembly was also changed. I'm not sure how it happened, however after uninstalling those other workbenches, the assembly remains broken. It's very important for users that want to adopt this a primary design platform that the solver be robust to changes of this nature, and these other glitches reveal that it is not in the present state.

### Full version info

```shell
OS: NixOS 24.11 (Vicuna) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.11, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.37

Note, Assembly3, Assembly4 and A2Plus were installed at the time this geometry was created. Though even after uninstalling them to the present state, the issue remains.
```

### Subproject(s) affected?

Assembly

### Anything else?

[4-bar-r02.zip](https://github.com/user-attachments/files/19405317/4-bar-r02.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-03-24:
try in 1.1dev

J-Dunn, 2025-03-25:
I have not tried assembly in quite some time but the reason I stopped using it was that once it got beyond assembling a few objects together , it got horribly slow. After that it got unstable. 

Furthermore, the topo instability meant you could bare change anything once it was part of an assembly.  That latter problem may be partially solved now.

drwho495, 2025-03-25:
> I have not tried assembly in quite some time but the reason I stopped using it was that once it got beyond assembling a few objects together , it got horribly slow. After that it got unstable. [...]

TNP mitigation is much better and there is a PR open to help with speed. Also joints now error with missing references so they won't be solved and cause a massive headache.

Zolko-123, 2025-04-10:
> I have not tried assembly in quite some time but the reason I stopped using it was that once it got beyond assembling a few objects together , it got horribly slow. After that it got unstable.

that is linked to the usage of a 3D solver : each constraint adds more entries to the solver, which necessarily becomes slower, to the point where it can get unsolvable. Assembly4's lack of solver is a feature, not a bug. That's what the Ondsel people never understood. There are many examples of assemblies with thousands of parts using Assembly4 :

https://forum.freecad.org/viewtopic.php?t=87934&start=10

FreeCAD's internal assembly WB is an evolutionary dead-end, it will never work for large assemblies.

drwho495, 2025-04-10:
> that is linked to the usage of a 3D solver : each constraint adds more entries to the solver, which necessarily becomes slower, to the point where it can get unsolvable. Assembly4's lack of solver [...]

Except for the fact that many commercial cad platforms successfully use solvers in their assemblies. The speed issue was half the solver and another half the time it takes to get edges/faces/vertices from a model. The latter has been improved with a PR that I made. Another fix is solving parts that have moved, and  any parts that are jointed to the part that has moved.

DavidLister, 2025-04-11:
> Except for the fact that many commercial cad platforms successfully use solvers in their assemblies. The speed issue was half the solver and another half the time it takes to get [...]

Exactly! I've used all of the major CAD packages and none are close to this unstable or slow. I have a feeling there are some insights that other packages use that aren't in FreeCAD. I think though looking at this blowup issue is a good thread to pull on, it's a sign of clearly pathological behavior and fixing it could help elsewhere.

Inventor/Fusion 360 is probably the worst commercial package in terms of assembly stability that I've used. They have a tendency to fall down other minima after significant move shift (resetting an animation is a classic case), however once the assembly breaks it doesn't blow up. The assemblies break easily, but they don't blow up.

The behavior this example file shows is special, after breaking each solver iteration takes it further from a sane position. For a stable solver, there should never be a case where the assembly explodes like this.

I don't know the intimate details of how the solver works, but I've skimmed the MbDTheory repo and the source and a few things stand out as potential sources. Some details are still lost on me, but the approach seems to run a kinematic simulation of the parts using physical units for mass and what appear to be some made up time units and some time step. I think part of the problem here could be the mass value as well as the time step, and possibly a missing drag term.

If the parts all have their mass determined by their volume, then assemblies mixing small and large parts may become unstable, as the forces resolved into small parts could cause massive displacements. Similarly, if the time step is too large, then dynamics will become unstable. Having all mass values set to the same number and having that adjustable may help. Setting the mass to the log of the mass could also help prevent large disparities while also maintain relative mass effects. Having mass and time step scaling adjustable could help to debug.

Damping is another concern, I see in the theory there is discussion of kinetic energy and inertia but no mention on damping. In the code however I don't see anything in the solver code that I can find (I'm certainly missing something, if someone can point me in the right direction that would be great!) that tracks inertia or kinetic energy in the simulation. If kinetic energy is part of the simulation though, it will need to be bled off with some sort of damping for the solution to be stable to larger perturbations. A concern I have is if it's implemented that the kinetic energy may start at zero during each interval without damping within the interval. If anyone can enlighten me here, I'd appreciate that a lot.

Will try 1.1dev shortly, what the expected change?

drwho495, 2025-04-11:
> Exactly! I've used all of the major CAD packages and none are close to this unstable or slow. I have a feeling there are some insights that other packages use that aren't in FreeCAD. I think though [...]

I agree, I think a lot of work needs to be done and sadly no one is currently maintaining ondselsolver

kwahoo2, 2025-07-26:
I can confirm it for the OP file and the latest weekly.
```
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42721 (Git) AppImage
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * freecad-xr-workbench.zip
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.94
  * kicadStepUpMod 11.4.4
```

```
12:38:33  3e-08 <Assembly> AssemblyObject.cpp(160): Solve failed: vector::_M_range_check: __n (which is 70) >= this->size() (which is 70)
12:38:34  0.983756 <Assembly> AssemblyObject.cpp(160): Solve failed: vector::_M_range_check: __n (which is 70) >= this->size() (which is 70)
12:38:34  Assembly : Ignoring bad solve, a grounded object (4-bar-r02#Fixed001) moved.
12:38:34  Assembly : Ignoring bad solve, a grounded object (4-bar-r02#Fixed001) moved.
12:38:34  Assembly : Ignoring bad solve, a grounded object (4-bar-r02#Fixed001) moved.
```

Also, tried to drag it, logs below:

[asmt-logs-4bar.zip](https://github.com/user-attachments/files/21443014/asmt-logs-4bar.zip)