FreeCAD/FreeCAD #20389: GEAR WB: Bevel Gears Aren't caling Correctly

fester225, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Post by [fester2251](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=780) » Sat Mar 22, 2025 7:23 pm

GEAR WB 1.3.0

I have two bevel gears, both 45 degrees, different numbers of teeth (36/9). So far I've found height, module, num_teeth, and pitch_angle to adjust. I'm leaving the pitch_angle the same for both, yet changing the module and num_teeth changes the actual pitch angle and tooth geometry in ways I hadn't expected.

I am including two bevel gears produced using the GEAR WB. They are identical except the number of teeth. The driving gear does not appear to be correct.

[BevelGear.pdf](https://github.com/user-attachments/files/19406498/BevelGear.pdf)

### Full version info

```shell
OS: macOS 15.3.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
* ThreadProfile 1.92.0
* freecad.gears 1.3.0
* AnimationFreeCAD 1.0.0
* pcb 6.2023.1
* offline-documentation 1.0.0-alpha
* kicadStepUpMod 11.4.4
* fasteners 0.5.35
* AirPlaneDesign 0.4.1
* Help 1.0.3
* Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Anything else?

GitHub is not accepting FreeCAD files. If you want the original file, tell me how  and where to upload it.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leuriato, 2025-03-23:
Hi there ! I believe you can upload your files here by putting them in a ZIP archive.

fester225, 2025-03-23:
[BevelGear.FCStd.zip](https://github.com/user-attachments/files/19406547/BevelGear.FCStd.zip)

luzpaz, 2025-03-23:
Probably the more appropriate place to post is at the https://github.com/looooo/freecad.gears/issues

closed 2025-03-23 by luzpaz

fester225, 2025-03-23:
> Probably the more appropriate place to post is at the https://github.com/looooo/freecad.gears/issues

Thank you. This is easily the best and most appropriate response I've gotten.  😊