FreeCAD/FreeCAD #20393: FreeCAD recovery is not working

uzytkownik, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When FreeCAD crashes I have an option to recover file. However if the backup is corrupted, FreeCAD seemed to corrupt file on recovery. Fortunately, it seemed to keep FCBak file so I was able to recover from backup by copying file.

(Sorry I forgot to save as I was too worried about a day loss of work)

### Full version info

```shell
OS: NixOS 25.05 (Warbler) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.11, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * kicadStepUpMod 11.5.2
  * fasteners 0.5.37
  * MeshRemodel 1.10.36
  * Assembly4.backup1714461781.0520585 0.50.6 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Can you provide an example file and steps to reproduce, maybe the corrupted file?
@hyarion do you kno where they are stored?

hyarion, 2025-03-26:
The files I've seen so far has always been cut off when writing a name (or was it label?)

Like:
`component name="` and then end of file.