FreeCAD/FreeCAD #20396: Model exported to STEP crash or produce empty objects

spkane, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have been having issues with multiple models recently when I try to export them to STEP files, even though they can be exported to STL and 3MF files, just fine. I have included two models here that exhibit various forms of these failures.

This first one appears to export fine, but results in an empty object:
[Balloons.FCStd.zip](https://github.com/user-attachments/files/19407052/Balloons.FCStd.zip)

This one completely crashes FreeCAD when I try to export it to a STEP file:
[Rewinder-Support-partial.FCStd.zip](https://github.com/user-attachments/files/19407062/Rewinder-Support-partial.FCStd.zip)

### Full version info

```shell
OS: macOS 15.3.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dracula.qss/OpenDark/Qt default
Installed mods: 
  * SearchBar 1.0.2
  * slic3r-tools
  * DynamicData 2.73.0
  * Curves.backup1741229329.291126 0.6.55 (Disabled)
  * Dracula 0.0.5
  * Render 2024.12.15
  * DynamicData.backup1741229335.1527839 2.71.0 (Disabled)
  * Defeaturing 1.2.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * lattice2 1.0.0
  * Curves.backup1721709375.3572009 0.6.41 (Disabled)
  * CubeMenu
  * 3D_Printing_Tools
  * ToolbarStyle
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.58
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-23:
I can reproduce the crash with the stable version, however, I failed to reproduce it with the unstable.

The exported model is incomplete.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40928 (Git)
Build date: 2025/03/22 22:27:10
Build type: Debug
Branch: master
Hash: c15e45e7e2f5c071d3ac8a097194e417b561d7b4
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

maxwxyz, 2025-03-24:
@spkane could you please test with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

spkane, 2025-03-25:
@maxwxyz I can't even get the current dev version macOS (ARM64) to start entirely. It loads up and then sits there with a spinning ball forever.

[image]

Syres916, 2025-03-25:
@spkane best to upgrade Dracula to version 0.0.6 to get it Qt6 compatible.

spkane, 2025-03-26:
> [@spkane](https://github.com/spkane) best to upgrade Dracula to version 0.0.6 to get it Qt6 compatible.

Thanks. Updating all the plugins before the upgrade fixed the booting issue.

spkane, 2025-03-26:
@maxwxyz I get the same (or very similar) behavior. 

[Balloons.FCStd.zip](https://github.com/user-attachments/files/19407052/Balloons.FCStd.zip) produces an empty object that crashes some STEP editors.

[Rewinder-Support-partial.FCStd.zip](https://github.com/user-attachments/files/19407062/Rewinder-Support-partial.FCStd.zip) crashes FreeCAD during the STEp export.

```
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: Dracula.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/151.5/2
Installed mods: 
  * SearchBar 1.0.2
  * slic3r-tools
  * DynamicData 2.73.0
  * Curves.backup1741229329.291126 0.6.55 (Disabled)
  * Dracula 0.0.6
  * Render 2024.12.15
  * DynamicData.backup1741229335.1527839 2.71.0 (Disabled)
  * Defeaturing 1.2.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * lattice2 1.0.0
  * Curves.backup1721709375.3572009 0.6.41 (Disabled)
  * CubeMenu
  * 3D_Printing_Tools
  * ToolbarStyle
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.60

```

spkane, 2025-04-01:
@maxwxyz The results are exactly the same when using the most recent build.

```
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: Dracula.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/151.5/2
Installed mods: 
  * SearchBar 1.0.2
  * slic3r-tools
  * DynamicData 2.73.0
  * Curves.backup1741229329.291126 0.6.55 (Disabled)
  * Dracula 0.0.6
  * Render 2024.12.15
  * DynamicData.backup1741229335.1527839 2.71.0 (Disabled)
  * Defeaturing 1.2.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * lattice2 1.0.0
  * Curves.backup1721709375.3572009 0.6.41 (Disabled)
  * CubeMenu
  * 3D_Printing_Tools
  * ToolbarStyle
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.60
```

In Safe mode, it is basically the same, but maybe not 100% the same. One of the files still causes FreeCAD to crash, but the one that used to create an empty STEP file now creates an "almost" empty STEP file. So, it is still very broken but a bit different.

```
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/151.5/2
```

FEA-eng, 2025-10-04:
Still relevant

The first file also freezes FreeCAD for me when reimporting.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2025-12-14:
@spkane Can you confirm if this issue is still relevant? For me, it behaves differently.

FEA-eng, 2026-09-15:
[This](https://github.com/FreeCAD/FreeCAD/issues/20396#issuecomment-4065468148) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

maxwxyz, 2026-09-15:
Closing, author inactive

closed 2026-09-15 by maxwxyz

luzpaz, 2026-09-15:
@maxwxyz file still crashes FreeCAD though? Is that not significant ?