FreeCAD/FreeCAD #20397: Weekly builds 40928 do not start

szdowk, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Weekly builds 40928 (FreeCAD_weekly-builds-40928-conda-Linux-x86_64-py311.AppImage) do not start:

FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

/tmp/.mount_FreeCACJBAej/AppRun: line 31: 11472 Aborted                 "${MAIN}" "$@"
Done.

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
Linux Slackware 15 x64.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

szdowk, 2025-03-23:
40655 worked without problems.

xtemp09, 2025-03-23:
Same on _KUbuntu 24.04.1 LTS_.

Smoerrebroed, 2025-03-23:
It looks like something QT is missing:

`FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
(qt.qpa.plugin) Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.`

theber-Co-60, 2025-03-23:
Same issue on ARCH and Tumbleweed.

JohnPatrickMc, 2025-03-23:
Same issue in Linux Mint

kaiwas, 2025-03-23:
Same issue on Manjaro Linux

thomasgi1, 2025-03-23:
Same issue on MacOS x86

kinghat, 2025-03-23:
Operating System: Fedora Linux 41
KDE Plasma Version: 6.3.3
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.2
Kernel Version: 6.13.7-200.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i7-5600U CPU @ 2.60GHz
Memory: 11.6 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 5500