FreeCAD/FreeCAD #20398: [TechDraw] Respect page order in tree when printing and numbering pages

j-hap, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently all pages in a document are auto numbered and printed in the order in which they are created (see https://github.com/FreeCAD/FreeCAD/blob/fabe25bd06e0a8f2bb26f3309ea91905f3a13ade/src/Mod/TechDraw/App/DrawTemplate.cpp#L107 https://github.com/FreeCAD/FreeCAD/blob/fabe25bd06e0a8f2bb26f3309ea91905f3a13ade/src/Mod/TechDraw/Gui/PagePrinter.cpp#L133-L134 https://github.com/FreeCAD/FreeCAD/blob/fabe25bd06e0a8f2bb26f3309ea91905f3a13ade/src/Mod/TechDraw/Gui/PagePrinter.cpp#L198-L199). I think it is more convenient that the tree order of the pages is respected in these actions, so a user could rearrange pages in a group and have their print order and page numbering updated accordingly. I already wrote the code to sort pages by tree order (currently only for printing), but wanted to check first how others think about it.
Also, if there already is a method to iterate over objects in a document in tree order, that would make it even easier, but I could not find one.

### Full version info

```shell
OS: Fedora Linux 41 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40609 (Git) Conda
Build date: 2025/03/09 13:41:26
Build type: Release
Branch: main
Hash: 844cceb04f0a82f55dee46445c4d57a7e20111ef
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/107.975/1
Installed mods: 
  * DynamicData 2.71.0
  * fasteners 0.5.37
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-03-23:
Seems like a good idea to me

j-hap, 2025-03-25:
Thanks for the feedback, I'll get going on a PR.

benj5378, 2025-03-25:
@WandererFan

j-hap, 2025-03-25:
I took a look at the auto numbering and I was mistaken, it is not done by creation order, but by ordering the page labels, so I can get the page numbering right if I just prefix the page number that I want (is that documented somewhere @pavltom ? ), but the print order could still be improved. I am hesitant to implement the same logic to sort by page labels though, because there would need to be a central place so sort any list of DocumentObjects by label to not duplicate the code.

I agree about the automatic numbering. The TemplateTextField should remember if it is set to autofill, which it currently does not, and then react so a trigger to regenerate the page number. In this case it's changing the label, but not only of the parent page, but any page, since that could change to order.

I did rewrite the `topologySort()` method of the `Document` class to output the tree order, which is different to what it returned before, but technically still a valid topology sort. So I got that going for me, which is nice...

@pavltom  I'd be happy to stick our heads together to come up with some improvement, if you're willing and have the time. You can find me on the FC discord under the same handle.

semhustej, 2026-08-18:
There is a related discussion in the forum: https://forum.freecad.org/viewtopic.php?t=103416