FreeCAD/FreeCAD #20402: Massive code rebuild on every make run

ickby, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD on my machine cannot be rebuild anymore without triggering heavy code compilation. This means that even without any changes to the code, just running "make" again after successful building, it takes many many minutes recompiling already build code. 

The output let me assume that the new code generation is responsible for this:
[image]

I'm not sure when it started, but must be recent. It is impossible to develop anything like this, as every recompilation takes way too long.

### Full version info

```shell
Main branch from today, 23.03.2025
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-03-23:
@ickby  You can temporarily reverse these changes: https://github.com/FreeCAD/FreeCAD/pull/20313

closed 2025-03-24 by hyarion

J-Dunn, 2025-03-26:
Just did pull and tested. Seem to fix the problem here. 
Thanks for the fix. That was a real time waster.

ickby, 2025-03-29:
Gesendet von Outlook für AndroidVon: marioalexis84 ***@***.***>Gesendet: Montag, März 24, 2025 12:49:43 AMAn: FreeCAD/FreeCAD ***@***.***>Cc: Stefan Tröger ***@***.***>; Mention ***@***.***>Betreff: Re: [FreeCAD/FreeCAD] Massive code rebuild on every make run (Issue
 #20402)

marioalexis84 left a comment ***@***.*** You can temporarily reverse these changes: #20313