FreeCAD/FreeCAD #20403: BIM - dialogues can be open multiple times

semhustej, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**
Some BIM dialogues are non-modal abd can be opened multiple times at once. I think this is an unwanted behavior, which can lead to user confusion.

Steps to reproduce:

- open new file
- open `BIM_ProjectManager` dialog
- invoke the `BIM_ProjectManager` again - second window with the same dialog opens

This happens with these tools:

- BIM_Setup
- BIM_ProjectManager
- BIM_IfcElements
- BIM_IfcQuantities
- BIM_IfcProperties
- BIM_Classification
- BIM_Layers

Proposed solution:
Upon invoking the tool for the second time, the first dialog should come into focus. There should be no second dialog opened.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-03-24:
[image]

Roy-043, 2025-06-24:
I was thinking of using `Gui.isCommandActive()`. But the results are strange. After closing the BIM_Setup dialog, the function still returns `True`. And for BIM_Layers it always returns `False` (even if the dialog is open).

Roy-043, 2025-06-24:
The proposed code has some issues. The windows are never removed from `allWindows()` it seems.
After opening BIM_Layers 3x and BIM_Setup 2x:
```
from PySide import QtGui
[w.objectName() for w in QtGui.QApplication.instance().allWindows()]
```
Returns:
`
['DialogWindow', 'DialogWindow', 'DialogWindow', '&ManageWindow', 'BIMSetupDialogWindow', 'BIMSetupDialogWindow', '&ManageWindow', '&UtilsWindow', 'QComboBoxPrivateContainerClassWindow', 'QWidgetClassWindow', 'Gui::MainWindowClassWindow']
`

Roy-043, 2025-06-24:
Maybe:
```python
def check_open_window(name):
    """raises a window with the given name if visible, and returns True. Otherwise, returns False"""

    for window in QtGui.QApplication.instance().allWindows():
        if window.objectName() == name and window.isVisible():
            window.raise_()
            return True
    return False
```