FreeCAD/FreeCAD #20408: Two-distance Chamfer not working on other end of two-sided, symmetrical model.

JDMils, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The issue is documented on the FC forums here:
https://forum.freecad.org/viewtopic.php?p=818069#p818069

One end of the model has been successfully chamfered in one hit however the other end will not chamfer in an expected format.

What we are seeing:
* Attempt to chamfer just the long straight edges results in a mis-matched & unwanted chamfer.

No matter which edge we choose, the final result is a mess. I do not think this is one of those chamfer/fillet issues as we already have a fully chamfered end on this model and we are performing the same action on the other end of the same model so it should chamfer the same way with the same result.

FC version 1.0.0 (rev 39109) responds the same way.
FC version 0.21.2 (rev 33771) responds the same way.

[image]

[image]

There is a fix however, you have to chamfer each edge on its own!? And the order in which you perform each chamfer is also important:
* Perform chamfer on Edge11 first.
* Perform a new chamfer on Edge8 next.
* Perform the next chamfer on Edge3 next.

The original FC model and my "fixed" model are both in the FC forum topic mentioned.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Logical/physical DPI: 96/106.57
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.58
  * fasteners 0.5.36
  * Gridfinity 0.8.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Please provide an example file.
@pieterhijma FYI

JDMils, 2025-03-26:
The original FC model and my "fixed" model are both in the FC forum topic mentioned.
I didn't think it was efficient to have multiple copies of the same files so I did not attach them here. Do you want me to attach them here?

maxwxyz, 2025-03-26:
Would be great, yes. Sometimes the last weeks the forum was not available and it is convenient for the developers to have all here.

JDMils, 2025-03-27:
Thank you. I zipped up 2 files and added them to the original post:
RodConnector.FCStd - The OP's original file provided in the forum post.
RodConnector_JDMils.FCStd - My "fix" where I had to chamfer each and every edge into their own chamfer object in order to get the end to chamfer correctly.