FreeCAD/FreeCAD #20414: TechDraw: Section task: 'Update Now' button should be 'Apply'

PaddleStroke, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

Currently the task has a 'Update Now' button. For consistency, shouldn't this be instead a 'Apply' button? @FreeCAD/design-working-group 

### Full version info

```shell
1.1
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-03-24:
I don't think so. Apply button is for closing the task, and the purpose of the current update button in not to close the task.

PaddleStroke, 2025-03-24:
No the apply button is not closing tasks. That's what the Ok button does. Apply button is acting like Ok but without closing. So basically the same as Update Now

benj5378, 2025-03-24:
Forum:
> IIRC, it used to be possible to set the values in the dialog without having the page updated until Apply was pressed. If the dialog now applies changes immediately, then the Apply button has no purpose.