FreeCAD/FreeCAD #20416: BIM: Unclone moves object

psommerfeld, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As done in attached project:
1. Create a BIM Window on BIM Wall. Call it 'original window'
2. Clone it. Call it 'cloned window'.
3. Selected cloned window, do Utils>Unclone.
Notice cloned window moves far away from original position (see screenshot)

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

BIM

### Anything else?

[image]

[image]

[unclone move.FCStd.zip](https://github.com/user-attachments/files/19427747/unclone.move.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

psommerfeld, 2025-03-24:
[image]

psommerfeld, 2025-03-24:
I just posted the exact same issue:
https://forum.freecad.org/viewtopic.php?t=95786
Must be a bug also.

> In my case, when I click unclone gives me some error messages and still behaves like cloned window. I mean, when I change original window height, uncloned window also changes its height
>

psommerfeld, 2025-03-24:
I created a new bug report for this under https://github.com/FreeCAD/FreeCAD/issues/20439

> I just posted the exact same issue: https://forum.freecad.org/viewtopic.php?t=95786 Must be a bug also.
> 
> > In my case, when I click unclone gives me some error messages and still behaves like cloned window. I mean, when I change original window height, uncloned window also changes its height

semhustej, 2025-03-25:
I did some testing and it seems to me that during the `Unclone` operation, the position values of the cloned window are added to the position values of the original window. Seems like a bug indeed.

https://github.com/user-attachments/assets/9a7b8b82-e1fd-45b9-8a1c-8985d1c69c2b

semhustej, 2025-03-25:
There is an older discussion about the unclone tool here: https://forum.freecad.org/viewtopic.php?t=38419

ErtugrulEge, 2025-10-03:
In this time, Utils->unclone doesn't work. Nothing happens when you select the cloned window and press the unclone.Please test it, if this is the case we can change the title such as BIM unclone doesn't work 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

semhustej, 2025-10-03:
> In this time, Utils->unclone doesn't work. Nothing happens when you select the cloned window and press the unclone.Please test it, if this is the case we can change the title such as BIM unclone doesn't work
> 

From my testing the unclone tool unclones the object but leaves the little clone icon in the project tree, so it may seem to user that the object is still a clone when in fact it is not. This is a new issue not related to the OP.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
  * UpdatedPythonPackages

```