FreeCAD/FreeCAD #20417: BIM: Property VIew>Data>"Clone Of" doesn't work

psommerfeld, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As in attached project:
1. Choose BIM Workbench
2. Select "cloned window" object
3. In Property View panel, choose Data pane
4. Press the "..." in the "Clone Of" field
5. Press Clear, then OK
6. Try to change the width of the window in Property View>Data>Width field
The following errors appear in the Report View:
`06:48:11  pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD 1.1.0dev.40176.app/Contents/Resources/Mod/BIM/ArchWall.py", line 521, in execute
    base = self.processSubShapes(obj,base,pl)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 1.1.0dev.40176.app/Contents/Resources/Mod/BIM/ArchComponent.py", line 781, in processSubShapes
    subvolume = o.getLinkedObject().Proxy.getSubVolume(o)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 1.1.0dev.40176.app/Contents/Resources/Mod/BIM/ArchWindow.py", line 594, in getSubVolume
    for w in base.Shape.Wires:
             ^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'Shape'
06:48:11  wall: 'NoneType' object has no attribute 'Shape'
`

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

None

### Anything else?

Is there a reason why "Clone Of" is even be an editable field?
If it is editable, then it seems to me the logical use would be to clear the clone relationship, ie. same functionality as Utils>Unclone.
But maybe there is a different intention.

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-03-24:
Also before error message  shows tons of   `<class 'AttributeError'>`
[image]

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/18390a6d04b94766d239ed2f5596593527bfa605

Clearing Clone Of no longer breaks the host wall, but it doesn't give the window a new base of its own.