FreeCAD/FreeCAD #20418: Techdraw: Arc dimension rendering appears broken

PaddleStroke, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

### Full version info

```shell
1.1 40800
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-24:
On a similar build but with Qt5 : 

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40516 (Git) Conda
Build type: Release
Hash: c44e29ee0bca1be812e1282106596a245b7c42f4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/88.8003

The bug is not here. So it is likely a Qt6-related bug.

Syres916, 2025-03-24:
https://forum.freecad.org/viewtopic.php?p=817951#p817951

benj5378, 2025-03-24:
I'm pretty sure the regression was made by me. I'll take a look

WandererFan, 2025-03-25:
Also affects Balloon leaders (and the bubble?).

PR #20431 (2025-03-26): TechDraw: arc dimensions should not have fill
Fixes #20418. Regression by commit 998f713c97fe870a845c566eb7597520ce26d6fd
commit c106e1fd (2025-03-26): TechDraw: arc dimensions should not have fill

Fixes #20418. Regression by commit 998f713c97fe870a845c566eb7597520ce26d6fd
closed 2025-03-26 by hyarion