FreeCAD/FreeCAD #20423: Problem with edge rounding. Video on youtube

lukashaxx, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad was the only program tested that failed to round the edges - is this a bug? I tested it and the problem is repeatable. The problem is shown in the link (from 1:20:40), the film also shows how other programs dealt with this problem:

https://youtu.be/LX4341s3sx4?si=UZrpRE7tpqmF_bCl&t=4779

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * InventorLoader 1.5.1
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Known issue of fillets, closing this in favor of https://github.com/FreeCAD/FreeCAD/issues/19703
@pieterhijma unless this is different.

closed 2025-03-26 by maxwxyz

pieterhijma, 2025-03-26:
I also think it's the same problem.