FreeCAD/FreeCAD #20425: BIM_Beam produces error when change of profile is undone

semhustej, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create BIM_Beam
- enter the edit mode of the beam profile
- select the different size of the profile from the list
- hit `OK` to confirm the change
- the profile is changed, everything is OK
- use `Std_Undo`
- the beam needs to be recomputed
- perform `Std_Refresh` 
- error shows at the profile level
```

15:06:32  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40504-conda-Windows-x86_64-py311\Mod\BIM\ArchProfile.py", line 159, in execute
    p1 = Vector(-obj.Width.Value/2,-obj.Height.Value/2,0)
                 ^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'Width'
15:06:32  UC152x152x23_: 'FeaturePython' object has no attribute 'Width'
```

See behavior in the video:

https://github.com/user-attachments/assets/ff2b48e1-8260-4dbf-83b1-464702333e51

I am able to reproduce the issue with `BIM_Column` too. 
I am able to reproduce the issue with FreeCAD 1.0.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Anything else?

A workaround is possible - enter the edit mode of the profile and select a new profile size.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-19:
Please retest after #22052 has been merged. AFAICT the PR also fixes this issue.

semhustej, 2025-06-27:
Tested with the latest dev release. The issue is fixed. Thank you.

I am closing this issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1

```