FreeCAD/FreeCAD #20432: 3mf export thumbnail _rels malformed

StefanBruens, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The package level relationships file (`/_rels/.rels`) should only contain a single thumbnail relation, valid for the whole package ("ZIP file").

Currently, there is a separate thumbnail for each part, but no thumbnail for the complete export:

[ ] There should be a single top-level thumbnail for the complete package/export.  
      [ ] Only this single thumbnail should be referenced from the root relationships file
[ ] Optionally, for each (sub-)part a separate thumbnail should be created
    [ ] These thumbnails must be listed in the corresponding relation file (e.g. `/3D/_rels/.rels`)
    [ ] These thumbnails must be referenced using a "thumbnail" attribute of the respective 3mf `<object />` element

See https://github.com/3MFConsortium/3mf-samples/blob/master/examples/production/cube_gears_prod.3mf for an example containing a package level and multiple object level thumbnails.

Attached: .FCStd input, .3mf export
[TwoBoxes.zip](https://github.com/user-attachments/files/19435888/TwoBoxes.zip)

### Full version info

```shell
OS: openSUSE Tumbleweed (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 5.15.16, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-09:
soft bump

luzpaz, 2026-01-19:
bump @StefanBruens