FreeCAD/FreeCAD #20437: Commit version in the FreeCad Window Title for Weeklies.

leoheck, 2025-03-24:
Can we have the commit version here in the title?

Right now it is this:

[image]

I can suggest something like this.
Instead of the `dev` string the commit `version` separated with a `-`

```
FreeCAD 1.1.0-40655
```

maxwxyz, 2025-05-11:
@oursland FYI

luzpaz, 2025-05-11:
This makes total sense. It can also really help with screencasts (if the user also captures the window title) it's easy to discern that it's a dev build and what specific revision it is.

oursland, 2025-05-12:
* [`src/Gui/MainWindow.cpp:MainWindow::setWindowTitle()`](https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/Gui/MainWindow.cpp#L2338) sets the title that it gets from `App::Application::getNameWithVersion()`
* [`src/App/Application.cpp:Application::getNameWithVersion()`](https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/App/Application.cpp#L1124) builds the version from the Config structure
* [`src/App/Application:Application::initConfig()`](https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/App/Application.cpp#L2527) initializes the Config structure with macros `FCVersion{Major,Minor,Point,Suffix}` that come from `<Build/Version.h>` that is created by CMake
* [`Version.h.cmake`](https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/src/Build/Version.h.cmake#L4) is built from the defaults in [`CMakeLists.txt`](https://github.com/FreeCAD/FreeCAD/blob/43b2a0e3017165b46c44b6fa4e00438ba4806ddf/CMakeLists.txt#L53)

For the weekly builds, this is likely not updated for the builds as desired.  The `FreeCAD-Bundle` repo uses [`make_version_file.py`](https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/make_version_file.py) to write out the Version.h with the git info.  I can work on getting a PR in to update this.

oursland, 2025-05-12:
* https://github.com/FreeCAD/FreeCAD/pull/21267
* https://github.com/FreeCAD/FreeCAD/pull/21268

luzpaz, 2025-05-15:
I hope we get a revision number too because we can quickly surmise how recent the build is compared to HEAD