FreeCAD/FreeCAD #20439: BIM: Changing size of uncloned object still changes old original

psommerfeld, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As seen in the video, I uncloned 'cloned window'. Then I changed its width. Notice the orignal that it had been cloned from also get its width changed.

Using the attached project:
1. Select 'cloned window' in Tree View.
2. Perform Utils>Unclone
3. Note the bug of the now-uncloned object moving (reported as [](https://github.com/FreeCAD/FreeCAD/issues/20416)).
4. Change the height of the now-uncloned object in the Property View panel.
Notice how the original object's width has also changed.

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

None

### Anything else?

I noticed that if the related objects are doors, I can change the Hinge Position of the newly-uncloned object and it properly does not affect the original.

https://github.com/user-attachments/assets/5fc3b753-25c4-4ff1-a015-d6a66165730b

[unclone move.FCStd.zip](https://github.com/user-attachments/files/19438504/unclone.move.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-25:
I can confirm the issue. 
I think the culprit is that the 'Uncloned' window still uses the same base object - the same sketch. The result is two independent window objects based on the same sketch. So if changing one of the window parameters (e.g. width) changes the underlaying sketch, all windows based on the sketch change.

The same happens when using the BIM_Copy tool.

This is quite unfortunate as some parameters are not connected with the original object and some are.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

yorikvanhavre, 2025-05-13:
This is a complex issue as there are MANY ways objects can be related parametrically in FreeCAD. Indeed after uncloning you have two independent windows, but still based on a same sketch.

Separating/duplicating the base sketch is not the aim of the unclone tool, at least currently. However, this tool could evolve in a more complex "break links" tool, and be able to do such deeper operations.