FreeCAD/FreeCAD #20448: BIM: slab has no Area property

furgo16, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is a feature request. Unlike wall objects, slab objects do not have an `Area` property. In fact, `obj.Shape` does not return the expected area, but as a workaround `obj.Base.Area` does.

The use case would be simply to be able to visualize the area of a given floor.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40958 (Git) Snap 1352
Build date: 2025/03/24 13:55:10
Build type: Release
Branch: main
Hash: 6b364bb169577519cd0b96a2e4e47d81807264d2
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-25:
Nevermind, there's the `HorizontalArea` for that.

yorikvanhavre, 2025-03-25:
I think in several cases, BIM objects shoudl have an "Area" property, that is linked to an (internal) area. For ex: for slabs, the Area = HorizontalArea, for walls, Area = VerticalArea, etc...