FreeCAD/FreeCAD #20454: projectToLine Bug

mylysddp, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I used point_new = hole_center.projectToLine(start_point, end_point) to project the point hole_center onto the line defined by start_point and end_point. However, the resulting projection point point_new has the exact same value as hole_center. What could be the reason for this? Is there an issue with how I’m calling it, or is there a bug in this method? Here, hole_center = (-1074, -1122, 382), start_point = (1140, 1190, 0), and end_point = (1140, 1190, 6700).

### Full version info

```shell
1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Please provide full about info and an example file including steps how to reproduce and what should be the expected outcome. TIA

luzpaz, 2025-04-02:
@mylysddp 👆

FEA-eng, 2025-10-04:
@mylysddp Can you share the file or exact steps to reproduce?

luzpaz, 2025-11-23:
No response from OP. Closing.

closed 2025-11-23 by luzpaz