FreeCAD/FreeCAD #20455: [PART] Create shape from mesh "sew tolerance" parameter is misleading and causes issues with shape refining

phaseloop, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Importing and refining STL meshes into FreeCAD shapes/solids gives a really bad user experience and usually not works well - due to one simple issue.

When running "Create shape from mesh" operation, so called "Sew tolerance" is disabled by default - which makes the code use default 10e-6 OCCT tolerance. Wiki says:

> Optionally check the Sew shape checkbox and specify a tolerance:
> 
>     This option is usually not needed. It is meant for mesh objects that are not watertight and have small gaps between edges. 

This is not exactly true. First of all - internally this parameter is just called "tolerance". And this is the difference how "Refine shape" operation works on imported mesh without tolerance setting and tolerance set to 0.1:

**Refine shape result without tolerance**

[image]

**Refine shape result with sew tolerance**

[image]

Lack of this being being enabled by default and misleading description is what was wrong with STL editing in FreeCAD since forever.

Example file attached.

[shape_from_mesh_examples.zip](https://github.com/user-attachments/files/19447442/shape_from_mesh_examples.zip)

### Full version info

```shell
All versions - tested on 1.1.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Thovthe, 2025-04-25:
### Three things:

This is more of a fix than a feature, isn't it?

It seems like there's some truth to the line in the wiki.  What, specifically, would you add to make it better?

It seems nice to be able to select the whole face polygon rather than individual tris but what are the real advantages of this change as it seems to leave the shape alone for non-planer faces?  Now I haven't tried to do mesh editing in freecad only blender so maybe this is obvious to others but for those who aren't, perhaps an explanation would be helpful.

phaseloop, 2025-04-25:
Well, I'll ask - what is the point of a "refine" operation which leaves the object exactly as it was before "refining"? :).

I tried to edit quite a few meshes from printables.com and it's frustrating. You can try to do an experiment: create a cube in STL, then import into FreeCAD. Then go to Part - select the cube top face and try to extrude it to add for example 10 cm.

Oops - you can't because you have 20 "partial" top faces instead of one.

panukettu, 2025-05-04:
As a CAD-noob I just concluded "Refine Shape" never helps and moved on to spend more time than I am willing to admit on manually reworking outputs. 

Now after testing with the sew shape on... well I am glad I stumbled upon this issue. Thank you @phaseloop