FreeCAD/FreeCAD #20456: STL export produces discontinuities in model (empty layers)

MariuszXC, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Attached model, when exported, produces some empty layers. 

[Freecad_project.zip](https://github.com/user-attachments/files/19447425/Freecad_project.zip)

This is the exported STL:

[STL.zip](https://github.com/user-attachments/files/19447489/STL.zip)

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * ThreadProfile 1.92.0
  * btl 0.9.9
```

### Subproject(s) affected?

None

### Anything else?

Result in slicer:

[image]

Slicer reports, that the STL has 368 open edges.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
How did you convert to STL, and what settings did you use in the Mesh WB?

Megamannen, 2025-03-30:
FreeCAD:
Notice the holes
[image]

Sliced (doesn't matter if I export to STL, 3mf or step)
Notice the lack of holes
[image]

FEA-eng, 2025-10-04:
@MariuszXC Since the issue seems to be visible only in a slicer, can you check if it's still relevant in the newest weekly build available [here](https://github.com/FreeCAD/FreeCAD/releases)?

FEA-eng, 2025-12-14:
@MariuszXC Can you confirm if this issue is still relevant?

t-gh-ctrl, 2026-02-25:
I have the same issue, whatever export method I try - a simple STL (/3mf) export or creating a mesh from a shape and exporting the mesh.
I tried a few different mesh options but all of them created empty layers (although not always the same).
I need to 3d print that part but I don't see how to "brute force" / work around this...

FreeCAD_weekly-2026.02.25-Linux-x86_64-py311.AppImage on fedora 42.

.FCStd showcasing the issue: [loft_issue.zip](https://github.com/user-attachments/files/25538699/loft_issue.zip)

Slicer:

[image]

JohnOCFII, 2026-02-25:
@t-gh-ctrl  Interesting.  It would appear the loft is created when it should not be, as it seems to be created with some crossed lines (from inner to outer part of the shade in this example).  I wonder if recreating your `opening_front` sketch (perhaps without the construction geometry?) might help the situation.

[image]

There are some warnings in the report window which might hint the in right direction:

```
08:57:02  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge1
08:57:02  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge2
08:57:02  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge2
08:57:02  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge6
08:57:41  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge1
08:57:41  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge2
08:57:41  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge2
08:57:41  <TopoShape> TopoShapeExpansion.cpp(1507): modified shape type Wire mismatch with Edge6
```

Whether exported as STL or STEP, there are definitely issues seen when the object is imported to a slicer:

[image]

```
OS: macOS 26.3
Architecture: arm64
Version: 1.2.0dev.45467 (Git)
Build date: 2026/02/24 18:54:53
Build type: Unknown
Branch: main
Hash: 34ae16cd01b179eb9e1801591276dbc5a38669b5
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * Telemetry 1.0.5
  * Curves workbench 0.6.70
```

t-gh-ctrl, 2026-02-25:
@JohnOCFII  - thank you for looking into this. It didn't occur to me to toggle transparency; indeed, the loft's lines are crossed. 

I removed all the construction lines in `opening_front` - no change. Ditto when removing constraints (I tried just in case).

Interestingly, if I change the diameter of the inner circle (the 69mm one) so that it becomes the outer circle, eg. by setting its diameter to 80mm, then the loft's lines aren't crossed and the STL export is fine.

Oddly, the order in which the circles are drawn seems to matter: when redrawing the sketch with the circles from scratch:
- drawing the inner circle first, then the outer one, and then applying the additive loft by selecting `opening_front` and adding the newly drawn sketch as a section: crossed loft lines
- but when drawing the outer circle first and then the inner one: the loft / STL export are OK