FreeCAD/FreeCAD #20460: Telemetry Add-on fails with older Python

JohnOCFII, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

My self-built pixi versions of FreeCAD work fine with the Telemetry Add-on, but I receive an error message with the most recent weekly build.  One difference noted is the version of Python is 3.13.0 for the self-built versions, and 3.11.11 for the most recent weekly build.  

```
08:35:31  During initialization the error "Posthog Python package is not installed: pip install posthog" occurred in /Users/osh/Library/Application Support/FreeCAD/Mod/FreeCAD-Telemetry/./InitGui.py
08:35:31  Please look into the log file for further information
```

### Full version info

```shell
Failing version info:

OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * FreeCAD-Telemetry 1.0.0
  * OpenTheme 2024.9.1
  * Render
  * fasteners 0.5.37

Working version info:

OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.40996 (Git) Conda
Build date: 2025/03/25 09:24:27
Build type: Release
Branch: main
Hash: 40ace3241a3b01db680944e649c3f0b04a60ff72
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * FreeCAD-Telemetry 1.0.0
  * OpenTheme 2024.9.1
  * Render
  * fasteners 0.5.37
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-03-25:
Did you clone this yourself, or install it with the Addon Manager?

JohnOCFII, 2025-03-25:
> Did you clone this yourself, or install it with the Addon Manager?

I installed with the Addon Manager (one of your early testers based on your Discord request).

chennes, 2025-03-25:
OK: what's happening is that when you installed it with the Addon Manager, it put the `posthog` package into a folder dedicated to the version of Python that you were using at the time. The new version of Python doesn't use the same package installation. If you uninstall and reinstall the addon with the FreeCAD that uses a different Python it should offer to install the `posthog` package again. I'll have to look at the Addon Manager's logic for handling Python version changes: I have some code in there, but I don't remember what it is.

JohnOCFII, 2025-03-25:
Interesting.  I used the Addon Manager to remove the Telemetry Add-on.  I then launched the latest weekly (the one built with Python 3.11.11) and attempted to install the Telemetry add-on.  First, I was surprised to see two different versions of the Telemetry Add-on showing, but nothing to indicate if they were actually different.

[image]

Then the installation failed:

[image]

I then launched back to the self-built version of FreeCAD with Python 3.13.0, and the Telemetry add-on installed correctly again.

chennes, 2025-03-25:
Do you still have the custom repo in your repo list?

JohnOCFII, 2025-03-26:
> Do you still have the custom repo in your repo list?

Yes, I did still have the custom repo configured.  I've just removed it and no longer have two entries in the Add-on Manager for Telemetry.

chennes, 2025-03-26:
OK good 👍 - so we are down to one bug, and one bug/feature request: the bug is that the Conda build's pip isn't running correctly, and the feature request (but really borderline bug) is that when you change versions of Python, the Addon Manager should detect that it doesn't have all the libraries the addons need and offer to install them.