FreeCAD/FreeCAD #20464: Using the scaling with `App.Vector(-1, -1, -1)` to mirror objects deforms the appearance of the object.

leoheck, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using the scaling with `App.Vector(-1, -1, -1)` to mirror objects deforms the appearance of the object.

The object on the left was mirrored to the left, and the colors were messed up.

[image]

Discussion: https://forum.freecad.org/viewtopic.php?p=818394#p818394

### Full version info

```shell
0.21, 1.0, 1.1-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-03-25:
This is probably due to due to inverted normals. Scaling with -1 should in general be avoided in 3d applications.

leoheck, 2025-03-25:
Do you know a way to fix this after the mirror happens?

hyarion, 2025-03-25:
You are basically turning it inside out. Why do you need to mirror fasteners instead of just placing them at the right places?

leoheck, 2025-03-25:
I am not mirroring fasteners; I am mirroring objects, and fasteners are one of them.
I have an object in one location, I have a plane, and then I am mirroring the object based on this plane.
But I am not doing this. This is a feature of a Workbench. 
It does not behave nicely, so I am trying to understand how I can fix it so it does not produce this issue.

hyarion, 2025-03-26:
Scaling is the wrong way to place objects. I've done it myself and felt pretty smart about a hack like that in the past, but it always comes and bites me in the butt with threads reversing and normals pointing the wrong direction.

I would argue that scaling with negative values are undefined at best and would be good to throw an exception when the user tries to do it.

It would be good if you file this as a bug in that workbench.

leoheck, 2025-03-26:
> Scaling is the wrong way to place objects. 

I see.

> It would be good if you file this as a bug in that workbench.

I did. But I am trying to fix it myself somehow.

Someone in the Freecad Forum (link in the first message of the post, above) mentioned it could be a bug, so I reported it here.

leoheck, 2025-03-26:
They mentioned this, and I agree.

> Yes, from my perspective. If you are scaling an object, this should not change the position of the light source.
Even if that scaling is negative, the programmer should account for this. If they cannot, or refuse to, then they should disallow negative scaling all together.

leoheck, 2025-03-26:
But other one said that:

> The mirroring has no impact on the light source. The light source is part of the viewer logic and objects of a document don't change anything there.

> The problem is basically a mathematical thing because using the scale factor of -1 for all coordinates one ends up with a 3x3 transformation matrix with the determinant of -1. This means that the orientation of the object is changed and surface normals do not point to the outside but to the inside.

> The effect is that the rendering shows the observed behaviour.

leoheck, 2025-03-30:
It seems that enabling the backlight in the settings can "improve" the visual quality of the resulting object. However, I am not sure this is the right fix.

hyarion, 2025-03-30:
No, the right fix is to avoid scaling using negative numbers.

leoheck, 2025-03-30:
This approach works on 0.21.3, but in 1.1 the back light is already enabled.

leoheck, 2025-03-30:
> No, the right fix is to avoid scaling using negative numbers.

Do you have an idea to help me fixing this?

jffmichi, 2025-04-12:
So we're holding it wrong? Scaling with negative numbers IS mirroring. If FreeCAD messes up the normals, that's a bug that should be fixed.

MiqCG, 2025-04-12:
PS: now I think that maybe I have to open a new issue for this, as it is related with the integrated assembly, and not (only) with Mod: Part and (only) with Topic: Color.
Let me know, @hyarion , and I will do it.

hyarion, 2025-04-12:
> If not, what is the correct way to obtain a mirrored part, [@hyarion](https://github.com/hyarion) ?

Yes this is a problem and I don't know a good way to do it currently:
* Part_Mirror would be one way, but that doesn't support mirroring App_Part
* Subshape binder + PartDesign_Mirrored would be another way, but PD_Mirrored doesn't support replacing original, only adding.

So I think your problem is a new issue: There's currently no way to mirror a part (that I know about) and this isn't only for assembly, another usecase would be when you want to create two similar mirrored parts, then there's no way I can find to do that.

MiqCG, 2025-04-12:
Not a new issue because it is already reported: 
> The issue is described [here](https://github.com/FreeCAD/FreeCAD/issues/14117) and [here](https://github.com/FreeCAD/FreeCAD/issues/5705) and they are still unsolved.

But, what about the possibility to use Scale = -1 that was working at least until March? (I don't have more appimages in between 39708 and 41127. I think it is the same or part of this original issue that has reported leoheck.
@leoheck , can you chech if, apart from the color, the mirroring works on your side? Please, test it with a non-symmetrical component. If that is the case, maybe you could update the title and the description.

leoheck, 2025-04-12:
@MiqCG there are some assymetric objects reported on the following link. They mirror as expected except by the color.

https://github.com/Zolko-123/FreeCAD_Assembly4/issues/552

If you follow the link above you can see that on previous versions of Freecad, enabling the background light in the settings used to fix this problem. But now, on 1.1 this is already on however the issue still persists.

leoheck, 2025-04-12:
See this  example here: [mirror_issue.FCStd.zip](https://github.com/user-attachments/files/19720742/mirror_issue.FCStd.zip)

Here, I have mirrored the object 2 times. 
[image]

leoheck, 2025-04-12:
@MiqCG do you need an appimage?
I have these saved:
```
FreeCAD_weekly-builds-41213-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41138-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41127-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40971-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40444-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40408-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-38553-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage
```

MiqCG, 2025-04-12:
I have downloaded your file but it involves Assembly4. This part that you are mirroring don't have a different form when mirrored and when not. On the other hand, we are using a different approach to mirror an object. Am I messing up talking about scale = -1 in here? How is `App.Vector(-1,-1,-1)` applied?

I have made more tests: if I link the Original_Arm.FCStd of my example to another file (without Assembly container) the issue arises too:
1. open the Original_Arm file
2. open a new file and save the file
3. link the object to the new file with `Make link`
4. apply scale = -1 in the properties panel (the mirrored part appears)
5. transforming (moving) the object make disappear the mirroring and the object turns blackSo, it is not exclusive of the integrated assembly WB.

@leoheck , can you download my example files and test the mirroring as described?
Depending on the result, I will ask you for appimages to find the moment the problem arouse.

FlachyJoe, 2025-04-12:
FYI the `Link` double-side rendering is checked here:
https://github.com/FreeCAD/FreeCAD/blob/e990b456b9db3952cf785cb38619622415b13dca/src/Gui/ViewProviderLink.cpp#L1848-L1864

leoheck, 2025-04-13:
@MiqCG Asm4 does this (I removed some irrelevant parts)

```python
class makeMirrorArray(makeExpressionArray):

    def _setupProperties(self, obj):
        obj.Count = 2
        obj.setExpression('Scaler', '1 - 2 * (Index % 2)')
        obj.setExpression('.Placer.Base', '.Placer.Rotation * minvert(.AxisPlacement) * .SourceObject.Placement.Base * -2 * (Index % 2)')
        obj.setExpression('.Placer.Rotation.Angle', '180 * (Index % 2)')

```

```python
class ExpressionArray(LinkArray):

    # do the calculation of the elements Placements
    def execute(self, obj):

        """ The placement is calculated relative to the axis placement
        Without Axis the Array is relative to the internal Z axis of the SourceObject."""

        sObj = obj.SourceObject
        # we only deal with objects that are in a parent container because we'll put the array there
        parent = sObj.getParentGeoFeatureGroup()
        
        if obj.Axis:
            obj.AxisPlacement = findAxisPlacement(*obj.Axis)
        else:
            obj.AxisPlacement = obj.SourceObject.Placement

        pmt1 = obj.AxisPlacement.inverse() * sObj.Placement

        placementList = []
        scaleList = []
        expDict = dict(obj.ExpressionEngine)
        evalList = _evalOrder(expDict)

        # calculate placement of each element
        for i in range(obj.Count):
            obj.Index = i
            for pn in evalList:
                ps = 'obj.'+pn.lstrip('.')
                nv = type(eval(ps))(obj.evalExpression(expDict[pn]))
                o,a = ps.rsplit('.',1)
                if a == 'Angle' and type(eval(o)) == App.Rotation:
                    nv = radians(nv)
                # must set entire rotation axis at once.
                # Related discussion: https://forum.freecad.org/viewtopic.php?t=73898
                if o.endswith('.Axis') and a in 'xyz':
                    axv = eval(o.replace('.Axis','.RawAxis'))
                    setattr(axv, a, nv)
                    exec(o + ' = axv')
                    continue
                exec(ps + ' = nv')
            placementList.append(obj.AxisPlacement * obj.Placer * pmt1)
            s = obj.Scaler
            scaleList.append(App.Vector(s, s, s)) #<---------------------- This is (-1, -1, -1) when obj.Scaler = -1
  
        # Resetting Index to 1 because we get more useful preview results 
        # in the expression editor
        obj.Index = 1
        if obj.ShowElement:
            for i in range(obj.Count):
                el = obj.ElementList[i]
                el.NoTouch = True
                el.Placement = placementList[i]
                el.ScaleVector = scaleList[i]
                el.setPropertyStatus('Placement',   'ReadOnly')
                el.setPropertyStatus('ScaleVector', 'ReadOnly')
                el.setPropertyStatus('Scale',       'ReadOnly')
                el.NoTouch = False
        else:
            obj.PlacementList = placementList
            obj.ScaleList = scaleList
        return

```

leoheck, 2025-04-13:
But I confirm that your example @MiqCG reproduces the issue.

Check here:
https://github.com/user-attachments/assets/e42fee89-9f0e-46df-b0c6-607652ac3d0f

MiqCG, 2025-04-13:
Sorry but I don't have enough programming skills to fully understand what the code that you sent really does.

Thanks for the testing. I will open another issue with this problem to not distort yours.

Can you give me access to some of this appimages?
FreeCAD_weekly-builds-40971-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40444-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40408-conda-Linux-x86_64-py311.AppImage

We can make it one by one, beginning with the oldest one. I just want to find the moment, to ease the finding of the problematic code change.

MiqCG, 2025-04-13:
New issue created: https://github.com/FreeCAD/FreeCAD/issues/20776
Thanks and sorry for the inconvenience.

leoheck, 2025-04-13:
@MiqCG, I think it is the same problem.

I've put my AppImages here.
https://www.dropbox.com/scl/fo/kk8ffuxc15ioj5l2b9r32/ABUUCxxigx1nlgVQMaPhdG8?rlkey=60qkhu7fwaxlh0m5p6jeaeonx&dl=0

It will take some time to upload them, but these are all the files you will find there
```
FreeCAD_weekly-builds-36438-2024-03-15-conda-Linux-x86_64-py310.AppImage
FreeCAD_weekly-builds-38553-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40408-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40444-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-40971-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41127-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41138-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41213-conda-Linux-x86_64-py311.AppImage
FreeCAD_weekly-builds-41232-conda-Linux-x86_64-py311.AppImage
```

luzpaz, 2025-05-09:
soft bump

leoheck, 2025-05-30:
It is the same as this duplicated issue here.
https://github.com/FreeCAD/FreeCAD/issues/20776#issuecomment-2920643217

leoheck, 2025-06-25:
https://github.com/FreeCAD/FreeCAD/issues/22170#comment-composer-heading

leoheck, 2026-03-06:
Thanks, I will check this.

FEA-eng, 2026-09-13:
I'm not sure how to reproduce it. Just mirroring like this doesn't affect the color.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.8854/Intel/Intel(R) UHD Graphics 620
```

jffmichi, 2026-09-13:
Iirc, mirroring something like a sphere gets you wrong light calculations. Have seen this a few days ago so still relevant.

luzpaz, 2026-09-19:
This is labeled 'Part' is that still correct ?