FreeCAD/FreeCAD #20467: TechDraw: face color for view doesn't work

benj5378, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

Setting the face color property on a TechDraw View doesn't work.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40974 (Git) Snap 1353
Build date: 2025/03/24 21:47:33
Build type: Release
Branch: main
Hash: 74493358dad6ce224d4d6597ef8f3bfb44a10a94
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: Darker.qss/Darker/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * Behave-Dark-Colors 0.1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-09:
Is this a regression ?

semhustej, 2025-05-09:
Yes, this seems like a regression to me. I am not able to reproduce in FreeCAD 1.0

Sample file (remove the .zip filename):

[face-color-test.FCStd.zip](https://github.com/user-attachments/files/20127225/face-color-test.FCStd.zip)

Open in FreeCAD 1.0

[image]

Open in FreeCAD 1.1 (Std_Refresh does not change anything)

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * BIM021 2021.12.0
  * Reporting
  * Telemetry 1.0.1

```

PR #21232 (2025-05-15): TechDraw: fix face color not working
Fixes #20467

## Issues

## Before and After Images