FreeCAD/FreeCAD #20470: DXF export working sometime only. Most of the time exported blank DXF

Praveenraj24, 2025-03-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Recently we are starting to use the FREECAD. The software designed as user friendly. We would like to appreciate that. When we export the DXF some times it working well. But most of the time exported as empty file.

### Full version info

```shell
Recently we are starting to use the FREECAD. The software designed as user friendly. We would like to appreciate that. When we export the DXF some times it working well. But most of the time exported as empty file.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-26:
Please provide full about Info and an example file.

Praveenraj24, 2025-03-26:
[TEST.zip](https://github.com/user-attachments/files/19464474/TEST.zip)

[CON_MOLEX_5044490607.zip](https://github.com/user-attachments/files/19464483/CON_MOLEX_5044490607.zip)

I have attached one screen recording and step file

Praveenraj24, 2025-03-26:
Most of time when we try to export the DXF from the step file, The exported DXF sized as 1KB and when we open the DXF it seems empty.

Roy-043, 2025-03-26:
FreeCAD currently cannot export 3D solids to DXF, at least not natively. The export option is called "Autodesk DXF 2D" for that reason. What type of DXF result do you expect/need?

Praveenraj24, 2025-03-27:
Please refer this attached another video file as I mentioned sometime its working. Please take needful action.

[TEST 2.zip](https://github.com/user-attachments/files/19483410/TEST.2.zip)

Praveenraj24, 2025-03-27:
I have checked another way I took two different files with same format. One of the step file DXF has been exported successfully. Another one step file seems not exported properly the file size as 1KB and seems empty. I have attached the files for your reference

[STEP FILES.zip](https://github.com/user-attachments/files/19483986/STEP.FILES.zip)

.

Roy-043, 2025-03-28:
Both step files can be imported and then exported to DXF. Attached the result for "CON_MOLEX_5013310207- NOT ABLE TO EXPORT.stp". Just like in the video I have used the legacy DXF exporter.

As explained, only 2D DXF export is supported ATM.

[5013310207wbm000_01.dxf.txt](https://github.com/user-attachments/files/19512050/5013310207wbm000_01.dxf.txt) (remove the .txt extension.)

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

maxwxyz, 2025-03-29:
@Roy-043 so is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/5708 ?

Roy-043, 2025-03-30:
This might be a duplicate. Maybe @Praveenraj24 can clarify. If they want 2D DXFs, then this issue may be due to something that is not clear about how DXF export works.

luzpaz, 2025-11-24:
> This might be a duplicate. Maybe [@Praveenraj24](https://github.com/Praveenraj24) can clarify. If they want 2D DXFs, then this issue may be due to something that is not clear about how DXF export works.

@Praveenraj24 👆