FreeCAD/FreeCAD #20472: Draft: Grid toggle button not always in sync with grid state

semhustej, 2025-03-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Priority: low

Steps to reproduce:

- set `Always show grid` and `Show the grid during commands` off in the settings
- restart FreeCAD
- start new file - grid is not visible
- start `Draft_Line` command
- turn grid on by clicking on the grid icon -> grid is visible, which is correct
- create a line by selecting first and second point
- when the command is ended, the grid is turned off (correct), but the grid button on the `Draft snap` toolbar and also status bar shows the grid turned on

Even though this does not limit functionality, it is a bit confusing for the user.

[image]

See behavior in the video:

https://github.com/user-attachments/assets/1079220d-9e84-4715-b8a5-32177578be70

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-26:
This has already been fixed with #20436.

closed 2025-03-26 by Roy-043

semhustej, 2025-03-26:
> This has already been fixed with [#20436](https://github.com/FreeCAD/FreeCAD/pull/20436).

What a coincidence :). Thank you.