FreeCAD/FreeCAD #20480: Deleting object crashes freecad.

leoheck, 2025-03-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For some reason, freecad is crashing when deleting one object.

Sample file:  [mirror_issue.FCStd.zip](https://github.com/user-attachments/files/19467010/mirror_issue.FCStd.zip)

I cannot delete any one of the selected files without crashing FreeCad.
Any idea on how can I remove these objects?

[image]

This is the crash itself:
```bash
➜ freecad ~/Downloads/mirror_issue.FCStd       

FreeCAD 1.1.0, Libs: 1.1.0devR40971 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x717f19045330]
#1  0x717f1b4170f0 in App::PropertyInteger::getValue() const from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADApp.so+0
#2  0x717f1c87dc2f in Gui::ViewProviderLink::onDelete(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x9f
#3  0x717f1c41c72d in StdCmdDelete::activated(int) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x60d
#4  0x717f1c405897 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x237
#5  0x717f1c405caf in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x12f
#6  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x717f1970a527]
#7  0x717f19f8d583 in QAction::triggered(bool) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0x43
#8  0x717f19f90972 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0xf2
#9  0x717f1c4671e0 in Gui::ShortcutManager::onTimer() from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x100
#10  0x717f1c467d66 in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x556
#11  0x717f1c46a477 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x97
#12  0x717f1974ceda in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x6a
#13  0x717f1a9a4132 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Widgets.so.6+0xc2
#14  0x717f1c3a64b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0xc9
#15  0x717f1974d29a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x16a
#16  0x717f19fa349a in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0x1ba
#17  0x717f19fa41a1 in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0x81
#18  0x717f19c2ae2d in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0x1ed
#19  0x717f19bc02b3 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Gui.so.6+0x153
#20  /tmp/.mount_FreeCAnDnFDb/usr/lib/qt6/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0x1154d) [0x717f1404754d]
#21  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x717f1970a527]
#22  0x717f18f665b0 in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/./libQt6DBus.so.6+0x40
#23  0x717f196d25db in QObject::event(QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x23b
#24  0x717f1a9a40f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Widgets.so.6+0x84
#25  0x717f1c3a64b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0xc9
#26  0x717f1974d29a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x16a
#27  0x717f1977c120 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x320
#28  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6(+0xf3604) [0x717f194f3604]
#29  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/./libglib-2.0.so.0(+0x5b1d5) [0x717f17dc71d5]
#30  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/./libglib-2.0.so.0(+0x5e377) [0x717f17dca377]
#31  /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x717f17dcaa8e]
#32  0x717f194efd0e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x6e
#33  0x717f197857fc in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x1ac
#34  0x717f19785c0a in QCoreApplication::exec() from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libQt6Core.so.6+0x9a
#35  0x717f1c2feae4 in Gui::Application::runApplication() from /tmp/.mount_FreeCAnDnFDb/usr/bin/../lib/libFreeCADGui.so+0x734
#36  /tmp/.mount_FreeCAnDnFDb/usr/bin/freecad(+0x6c50) [0x5e74c8a24c50]
#37  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x717f1902a1ca]
#38  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x717f1902a28b]
#39  /tmp/.mount_FreeCAnDnFDb/usr/bin/freecad(+0x6fd1) [0x5e74c8a24fd1]

```

This does not happen on 1.0 and 0.21

[... 84 lines cut ...]
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-26:
Look at the seg fault in https://github.com/FreeCAD/FreeCAD/pull/20482 I think this is the same fix

leoheck, 2025-03-26:
Indeed, since it is related with Link Array.

closed 2025-03-26 by hyarion

kaiwas, 2025-03-29:
A temporary solution (until a program update is available) is to downgrade the problematic object and delete the result.

leoheck, 2025-03-29:
Thanks @Kaias. How do I downgrande the object?

kaiwas, 2025-03-29:
> Thanks [@Kaias](https://github.com/Kaias). How do I downgrande the object?

In the BIM workbench. Tool with an arrow pointing down.

https://github.com/user-attachments/assets/a53d718e-0cc0-4e70-86ae-865133848ca9

leoheck, 2025-03-29:
@kaiwas downgrading the object gives me this:

```
13:08:16  Traceback (most recent call last):
  File "/home/lheck/Applications/squashfs-root/usr/Mod/Draft/draftutils/todo.py", line 165, in doTasks
    Gui.doCommand(string)
  File "<string>", line 1, in <module>
  File "/home/lheck/Applications/squashfs-root/usr/Mod/Draft/draftfunctions/downgrade.py", line 340, in downgrade
    if obj.Shape.ShapeType != "Edge":
       ^^^^^^^^^^^^^^^^^^^
Base.FreeCADError: cannot determine type of null shape

13:08:16  ToDo.doTasks, Unexpected error:
<class 'Base.FreeCADError'>
in ['_objs_ = Draft.downgrade(FreeCADGui.Selection.getSelection(), delete=True)']
```

It does not work and freecad still crashes after trying to delete the object.

kaiwas, 2025-03-29:
> It does not work and freecad still crashes after trying to delete the object.

Sadly, we had to try.
So we wait for the update. I also encountered this problem when I was working on link array. But in my case, downgrading helped.

leoheck, 2025-03-30:
@kaiwas I was able to delete these objects in the Python console by doing:
1. Selecting the object.
1. Sending object to the Python console with `Ctrl+Shift+P`.
2. And doing `doc.removeObject(obj.Name)`