FreeCAD/FreeCAD #20485: Crash when closing tabs of multiple files

leoheck, 2025-03-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was closing multiple files with `Ctrl+W`.
The files were linked in an assembly.

While I was closing them, FreeCAD crashed.
Perhaps this happens since some job or computation is not done in another file, while the other is being closed.

These are the last things I saw on the command line after the crash.

```
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
LCS_Seal: No planar face in AttachEngine3D::calculateAttachedPlacement()!
<Exception> No planar face in AttachEngine3D::calculateAttachedPlacement()!
LCS_Seal: No planar face in AttachEngine3D::calculateAttachedPlacement()!
<Exception> No planar face in AttachEngine3D::calculateAttachedPlacement()!
LCS_Seal: No planar face in AttachEngine3D::calculateAttachedPlacement()!
<Exception> No planar face in AttachEngine3D::calculateAttachedPlacement()!
LCS_Seal: No planar face in AttachEngine3D::calculateAttachedPlacement()!
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
Program received signal SIGSEGV, Segmentation fault.
$HOME/Applications/squashfs-root/AppRun: line 31: 69235 Segmentation fault      (core dumped) "${MAIN}" "$@"
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

None

### Anything else?

To give more context, launching freecad after the crash, I see this:

```
15:29:21  Ignore recovery file  "${HOME}/.cache/FreeCAD/Cache/FreeCAD_Doc_006783d9-bebe-4303-8fe5-289c34cd66fb_b09ee2_297894/fc_recovery_file.fcstd"  because it is older than the project file "[OMITTED_FILE_PATH]/Parts/PartA.FCStd" 

15:29:21  Ignore recovery file  "${HOME}/.cache/FreeCAD/Cache/FreeCAD_Doc_ebd4c553-fc23-4e3c-83ed-5fbb0b181837_193fe7_297894/fc_recovery_file.fcstd"  because it is older than the project file "[OMITTED_FILE_PATH]/Parts/PartB.FCStd" 

15:29:21  Ignore recovery file  "${HOME}/.cache/FreeCAD/Cache/FreeCAD_Doc_fbdd77ef-b712-42c7-967b-a5b2b2ddf6cd_871e93_297894/fc_recovery_file.fcstd"  because it is older than the project file "[OMITTED_FILE_PATH]/Checks/PartC.FCStd" 
``` 

The PartA and PartB are in use (linked) inside the PartC.
So it was probably a crash related to closing a file that still has computations needed to be done on another file or something similar.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-29:
@PaddleStroke FYI

PaddleStroke, 2025-03-30:
Someone has been making changes to the partially loaded file mechanismes. After that we had a bug that all partially loaded files would open a tab anyway recently.
I cannot find this issue anymore but I think it could be related / regression perhaps.

leoheck, 2025-03-30:
Yeah @PaddleStroke this a horrible issue.

maxwxyz, 2025-07-05:
Thanks! Closing then.

closed 2025-07-05 by maxwxyz