FreeCAD/FreeCAD #20486: Feature - Separate the Report View content for each opened file.

leoheck, 2025-03-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

So right now the Report View just shows a bunch of messages sequentially. But working with multiple files make things messy.

So, when a file/tab is selected or when a file is opened, it could print some text to show we are working on a different file... or even better... each file could show different content on the Report view.

However, simply printing something when the focus changes from one file to another would be enough to also have some sort of separation on the command line. 

### Full version info

```shell
0.21, 1.0, 1.1-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-04-11:
How essential is this feature? _FreeCAD_ generates a log file that applies to all projects rather than being specific to individual ones. _Report View_ is designed to follow a similar approach.

> But working with multiple files make things messy.

_Report View_ per file will make things messy.

xtemp09, 2025-04-11:
I typically work on just one project at a time, and, occasionally, two. I understand that many users who manage multiple projects often have several opened simultaneously, usually around six or nine. It will create complete and utter mess if it adds 6-9 report views.

luzpaz, 2025-05-07:
Seems related to  
* https://github.com/FreeCAD/FreeCAD/issues/11914