FreeCAD/FreeCAD #20490: [UI/UX] Improved input fields. Automatically trim zero values ​​after the dot.

kaiwas, 2025-03-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, we can set "Nuber of dicemals" to 1 or more characters. This allows us to maintain the desired precision. (Unfortunately, there is no option to set 0)

But even if we have an integer set in the input field, we see extra zeros after the dot according to the setting.

This causes inconvenience when we need to change the last integer values ​​in the field.
For example, from 1250 to 1200. The field displays 1200.0 and we need to delete ",0" before we can change 50 to 00.

Video demonstration.

https://github.com/user-attachments/assets/0de8db91-c4ac-402d-ba05-f7251e0b39ad

In addition, this would help to eliminate the problem when "Nuber of dicemals" is applied at the time of object creation. At the end of the video, I change the parameter, but the values ​​of the previously created wall display the number of characters without taking this setting into account.

My suggestion. Implement cutting off extra zeros in values ​​after the dot. For example, if "Nuber of dicemals = 3", and the parameter 123.400 in the fields displays 123.4
at 123.000 displays 123
at 123.450 displays 123.45

The problem manifests itself not only in the DRAFT/BIM workbenches, it is present in the entire program as a whole. There is no need to display zero values. They only interfere.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda AppImage
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-28:
@FreeCAD/cad-working-group @FreeCAD/design-working-group what is your opinion on that?

pierreporte, 2025-03-28:
Sounds fine

MisterMakerNL, 2025-03-28:
Trailing zero's seem like a bug too me. It has no use, and can only be confusing with different notations.
Example: 1000 >1.000>1,000