FreeCAD/FreeCAD #20498: [BIM] Program crash when merging walls.

kaiwas, 2025-03-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Most often, the problem occurs if you combine (merge) three or more walls at the same time - there is a high probability of a crash. Usually 2-3 attempts are enough.

The crash also occurs for two walls, but much less often.

Video demonstration.

https://github.com/user-attachments/assets/218c8469-11b8-4d81-b91b-1d93cf69a7d6

Tested in safe mode too. 
The position of the walls does not matter.

There is no problem in version 1.0.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda AppImage
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC
  * OpenTheme 2024.9.1
  * freecad.gears 1.3.0
  * PitchedRoof
  * Test
  * sheetmetal 0.7.22
  * Road 2025.2.22
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-28:
I am not able to reproduce the crash. I tried to repeat `Merge Walls`  7 times. The errors on the screenshot are not related to the `Merge Walls` tool.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

kaiwas, 2025-03-28:
I checked on two computers with Manjaro Linux and everywhere this problem. I tried to reset the settings. It does not help.

semhustej, 2025-03-28:
Let's see if another Linux user can reproduce. Also - Windows weekly is I think 14 days behind the Linux builds.

xtemp09, 2025-03-28:
> I checked on two computers with Manjaro Linux and everywhere this problem. I tried to reset the settings. It does not help.

Could you provide the backtrace?

kaiwas, 2025-03-28:
> Here is the backtrace.

Here is the backtrace.

jonzirk76, 2025-03-29:
This is happening on my build as well.  Sometimes with the first merge.

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40724 (Git) Conda
Build date: 2025/03/20 10:05:03
Build type: Release
Branch: fixTheThing
Hash: 9b5b4621370ccb64120f6434a572970b8375a2ee
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/163.195/1

jonzirk76, 2025-03-29:
Happening on my system too, also in safe mode

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

Syres916, 2025-03-29:
So it's only on Qt6.x builds and has been around for months based on the feedback.

xtemp09, 2025-03-30:
The cause of the crash is absence of testing for `nullptr` of a pointer at

https://github.com/FreeCAD/FreeCAD/blob/e7156aa32608e7d06eefb2fe7fd0dd8d1aa7f138/src/Gui/Tree.cpp#L5462

For some reason, `getTree()` is `nullptr`, perhaps, because the object is not attached to the tree.

Syres916, 2025-03-30:
@xtemp09 [this](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Qt6_BIM_Merge_Walls_Fix) works on Qt6.7.2

Syres916, 2025-03-30:
@maxwxyz / @luzpaz could the _OS: Linux_ label please be removed and the _3rd Party: Qt6_ be added as it doesn't matter which platform the Qt6 build is run on, the issue remains.

Syres916, 2025-03-30:
@xtemp09 I'm always for getting the job properly rather than leaving a hack lying around, cheers.

xtemp09, 2025-03-30:
@Syres916, I think this hack is more good-looking:

https://github.com/FreeCAD/FreeCAD/compare/main...xtemp09:FreeCAD:iss-20498-BIM

Syres916, 2025-03-30:
Just sticking with the root cause and using some very crude Console output, it looks like using a Qt6 build that `void DocumentObjectItem::testStatus(bool resetStatus, QIcon& icon1, QIcon& icon2)` is being triggered much more than using a Qt5 build, based on a 3 Wall merge.

Qt5 output:

```
18:10:17  Unnamed#WallTrace
18:10:17  Tree is available
18:10:17  Unnamed#WallTrace001
18:10:17  Tree is available
18:10:17  Sketcher::setUpSketch()-T:7.5965e-05
18:10:17  Sketcher::setUpSketch()-T:4.1579e-05
18:10:17  Sketcher::Solve()-DogLeg-T:1.4747e-05
18:10:17  Unnamed#WallTrace002
18:10:17  Unnamed#WallTrace001
18:10:17  Unnamed#Wall002
18:10:17  Unnamed#WallTrace
```

Qt6 output:

```
18:05:17  Unnamed#Wall002
18:05:17  Unnamed#WallTrace002
18:05:17  Tree is available
18:05:17  Unnamed#Wall001
18:05:17  Unnamed#WallTrace001
18:05:17  Unnamed#WallTrace
18:05:17  Tree is NOT available
18:05:17  Unnamed#WallTrace
18:05:17  Tree is available
18:05:17  Unnamed#WallTrace001
18:05:17  Tree is available
18:05:17  Sketcher::setUpSketch()-T:6.1054e-05
18:05:17  Sketcher::setUpSketch()-T:2.3667e-05
18:05:17  Sketcher::Solve()-DogLeg-T:1.3342e-05
18:05:17  Unnamed#Wall002
18:05:17  Unnamed#WallTrace002
18:05:17  Tree is available
18:05:17  Unnamed#WallTrace001
18:05:17  Unnamed#WallTrace
```