FreeCAD/FreeCAD #20505: Build configuration confusion

3x380V, 2025-03-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Building for QT5:
```
$ cmake ../FreeCAD
-- Using ccache found at: /usr/bin/ccache
-- Compiler: GNU, version: 12.2.0
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.25.1
-- Compiling with Qt 5
-- Checking for connection to GitHub...
-- GitHub connection established for FetchContent
-- Module support is disabled.
-- Version: 9.1.0
-- Build type: 
-- CXX_STANDARD: 20
-- Required features: cxx_variadic_templates
-- fmt was downloaded using FetchContent into /home/ladis/src/FreeCAD-build-qt5/_deps/fmt-src
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter Development Development.Module Development.Embed 
-- Eigen found (include: /usr/include/eigen3, version: 3.4.0)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.65.0") found components: system filesystem date_time iostreams serialization 
-- Found flann version 1.9.2
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- Found Qhull version 8.0.2
-- OpenNI found (version: 1.5.4.0, include: /usr/include/ni, lib: /usr/lib/libOpenNI.so;libusb::libusb)
-- OpenNI2 found (version: 2.2.0.33, include: /usr/include/openni2, lib: /usr/lib/x86_64-linux-gnu/libOpenNI2.so;libusb::libusb)
** WARNING ** io features related to pcap will be disabled
-- Eigen found (include: /usr/include/eigen3, version: 3.4.0)
-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_2D
-- looking for PCL_FEATURES
-- looking for PCL_SURFACE
-- looking for PCL_IO
-- looking for PCL_FILTERS
-- looking for PCL_GEOMETRY
-- looking for PCL_ML
-- looking for PCL_SEGMENTATION
-- looking for PCL_SAMPLE_CONSENSUS
-- Found pybind11: /usr/include (found version "2.10.3")
-- PyCXX found:
--   Headers:  /home/ladis/src/FreeCAD/src/3rdParty/PyCXX
--   Sources:  /home/ladis/src/FreeCAD/src/3rdParty/PyCXX/CXX
--   Version:  7.1.9
-- -- Found OpenCASCADE version: 7.6.3
-- -- OpenCASCADE include directory: /usr/include/opencascade
-- -- OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;IOMPIParallel;ParallelMPI;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so  found components: OpenGL GLX 
-- Found HDF5: hdf5;mpi;/usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so;/usr/lib/x86_64-linux-gnu/libcrypto.so;/usr/lib/x86_64-linux-gnu/libcurl.so;/usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libz.so;/usr/lib/x86_64-linux-gnu/libdl.a;/usr/lib/x86_64-linux-gnu/libm.so (found version "1.10.8") found components: C HL 
-- Check for medfile (libmed and libmedc) ...
-- We guess that libmed was built using hdf5-openmpi version
-- Set up to compile with Qt 5.15.8
-- Checking Pivy version by importing it in a Python program...
-- Found Pivy 0.6.8
-- Checking Pivy Coin3D version by importing it in a Python program...
-- Found Pivy Coin3D SIM Coin 4.0.0
-- Shiboken2Config: Using default python: .cpython-311-x86_64-linux-gnu
[... 606 lines cut ...]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41028 (Git)
Build date: 2025/03/28 06:43:15
Build type: Unknown
Branch: main
Hash: 8c6e77ee972ef1c99d2ab6a84c7ef4b04cc1bf07
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/129.483/1
Installed mods: 
  * fasteners 0.5.36
  * Assembly4 0.50.13
  * sheetmetal 0.7.21
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-03-28:
I confirm `./bin/FreeCADCmd -t 0` crash when `./bin/FreeCADCmd` run well.

3x380V, 2025-03-28:
So, removing shiboken6m shiboken6_generator and pyside6 from pip packages makes `./bin/FreeCADCmd -t 0` work again (no segfault).

As for the second part:
```
ImportError: /usr/local/lib/python3.11/dist-packages/PySide6/Qt/lib/libQt6Gui.so.6: undefined symbol: _ZTVNSt3pmr25monotonic_buffer_resourceE, version Qt_6
```
is probably related to different compiler used to build Debian package and pip packages. Will investigate more.

3x380V, 2025-11-21:
Sorry for the delay. Except for CAM tests still failing here (which is the same for the last year or so) current master configures, compiles and runs built against both Qt5 and Qt6.

closed 2025-11-21 by 3x380V