FreeCAD/FreeCAD #20511: Cannot delete object that is an Attachment of other.

leoheck, 2025-03-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have an `Origin006` that appeared somehow.
Then, by some mistake, it is being used as the attachment mode of the `Sketch492`

Sample file:  [Origin_issue.FCStd.zip](https://github.com/user-attachments/files/19510368/Origin_issue.FCStd.zip)

However, I cannot delete it. 
It shows this popup but the `Yes` does not delete it.

[image]

Can I delete it or not?
I think we should be able to delete objects that are linked. 
It can complain, but it should allow me to delete without having to change the attachment of the Sketch first.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-03-30:
I was able to delete these objects in the Python console by doing:
1. Selecting the object.
1. Sending object to the Python console with `Ctrl+Shift+P`.
2. And doing `doc.removeObject(obj.Name)`

FEA-eng, 2025-10-04:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

leoheck, 2025-10-04:
The delete could have a forced mode to delete things no matter the consequences.

FEA-eng, 2026-10-25:
It still shows the warning, but allows me to delete that object.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```