FreeCAD/FreeCAD #20514: Freecad 1.0.0 becoming unresponsive when opening a new file or clicking on Parametric Part button

prashanthsmurthy, 2025-03-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am unable to create a new file using either the File -> New (Ctrl + N) option or by clicking Parametric Part on the start screen.  Here are the steps I followed

1. Install FreeCad 1.0 from the download link on Windows 11 Pro.  I have tried both the installer and the .7z portable versions with exactly the same problem
2. Delete C:\Users\prashanth\AppData\Roaming\FreeCAD folder to ensure its a brand new install and no previous version info affects it
3. Open FreeCad from the Start menu.  This opens the initial setting to choose one of the UI.  Doesn't matter if I select the Light theme or the Dark theme, the problem occurs in both
4. Create a new file using either File -> New or Ctrl+N or click on the Parametric Part option

The app become unresponsive and I am attaching a screenshot of the same:

[image]

[image]

I have tried to restart the application in Safe Mode and try the same and the results are the same.  I was quite happy with 0.21.1 version where I had no such problems, but now I have uninstalled it and unable to use the new version either.

Regards,
Prashanth

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

I have included the steps I have tried and the screenshots of the frozen screen.  I will be happy to help with any additional information you may want, but do let me know the details of what and how to extract information and I will be happy to be of help

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-03-29:
Perhaps, there are ways to obtain information even in this case.

- Can _FreeCAD_ open an existing file?

- If _FreeCAD_ is launched with `-l` option, it creates a log file. [^1] Could you reproduce the steps and upload the log file?

[^1]: https://wiki.freecad.org/Debugging

prashanthsmurthy, 2025-03-29:
> Perhaps, there are ways to obtain information even in this case.
> 
> * Can _FreeCAD_ open an existing file?

No, its the same behaviour of FreeCad not responding

> * If _FreeCAD_ is launched with `-l` option, it creates a log file. [1](#user-content-fn-1-1de4f6a735d398f6653402e2177fd474) Could you reproduce the steps and upload the log file?
> 

I am attaching the FreeCAD.log file that was created at C:\Users\prashanth\AppData\Roaming\FreeCAD\FreeCAD.log

[FreeCAD.log](https://github.com/user-attachments/files/19522875/FreeCAD.log)

> ## Footnotes
> 1. https://wiki.freecad.org/Debugging [↩](#user-content-fnref-1-1de4f6a735d398f6653402e2177fd474)

xtemp09, 2025-03-30:
I don't know the cause of the freeze, but here are my observations:

The _PATH_ contains:

- Python 3.12
- Python 3.9
- .
- Many other things, some of them more than once, for example, _OpenSSH_.

I would first clean _PATH_ before launching _FreeCAD_ or launch _FreeCAD_ with _PATH_ containing only standard things, like system32. [This link](https://wiki.freecad.org/Frequently_asked_questions#FreeCAD_crashes_on_creating_a_new_document_or_opening_a_file) recommends to launch the program from a command-line, could you start _FreeCAD_ from a terminal to see if it outputs something?

prashanthsmurthy, 2025-03-30:
Hi xtemp09,

I tried to clean a bit of the path, there were some paths getting applied at the system level and user level and then there were cases like C:\Windows and C:\WINDOWS (different cases) getting added.  There were also a few %SYSTEMPATH% which would point to the same C:\Windows.  I hope I have cleaned enough.  I don't have Python 3.9, only Python 3.12 that I installed quite some time ago.  I am not much of a python developer, but more of a Java / .Net guy!

[FreeCAD - Verbose Crash.log](https://github.com/user-attachments/files/19525202/FreeCAD.-.Verbose.Crash.log)
I ran it from command line and even though I used --verbose, I did not see anything printed out

"C:\Program Files\FreeCAD 1.0\bin>.\freecad.exe -l --verbose

C:\Program Files\FreeCAD 1.0\bin>"

[FreeCAD - Verbose Nooperation.log](https://github.com/user-attachments/files/19525219/FreeCAD.-.Verbose.Nooperation.log)
The above one is the same command that I run in command line with the --verbose option, but I simply close the UI without clicking New or Parameteric model.  Then there is no crash or unresponsiveness and the program terminates correctly.  Again, no more output seen on console.

"C:\Program Files\FreeCAD 1.0\bin>.\freecad.exe -l --verbose

C:\Program Files\FreeCAD 1.0\bin>.\freecad.exe -l --verbose

C:\Program Files\FreeCAD 1.0\bin>"

And just in case, I am using a normal command prompt (running as user) and not powershell

xtemp09, 2025-03-30:
Well, _PATH_ still contains too much of things that must not be there, like ".", Python39.

There is a way to find out the place in which _FreeCAD_ gets frozen. One has to launch _FreeCAD_ in a debugger, like _gdb_ or _lldb_, when _FreeCAD_ freezes, press _Ctrl+C_ and take a look at the backtrace. This will show the function _FreeCAD_ is trying to invoke.

prashanthsmurthy, 2025-04-01:
Really sorry to get back late xtemp09.  The "." is there for a purpose, to put the current directory in the path.  Python39 doesn't exist even if it appears in the path, it won't make any difference.  I can definitely try to run it in gdb as I do have mingw-gdb installed, but I haven't tried debugging in over 20 years, so definitely quite rusty on that.  If you can guide me, I can run the program in the debugger and share the findings.  If you want some sort of screen sharing, can try to work that out too.  I live in India (GMT + 5:30), just in case you want to know.  Do let me know the commands to try out or a convenient time for you.

In the meanwhile, I tried to run gdb -ex run freecad.  I clicked on Parameteric and nothing happened, when I killed the process, it killed the mingw process itself, so I couldn't find much.  I also tried linking to an existing process using gdb --pid=<PID of Freecad>, but here again, I saw the program end, but don't know how to take core-dumps or stack traces.

xtemp09, 2025-04-01:
In command line:

```
gdb FreeCAD.exe
```

or

```
lldb FreeCAD.exe
```

in the command line of _gdb_ or _lldb_:

```
(gdb) run
```

When _FreeCAD_ freezes, in the command line one has to press _Ctrl+C_ and then enter "_bt_":

```
(gdb) bt
```

The backtrace will show the function _FreeCAD_ is stuck on.

prashanthsmurthy, 2025-04-01:
prashanth@MyDesktop MINGW64 ~
$ cd /c/Program\ Files/FreeCAD\ 1.0/bin

prashanth@MyDesktop MINGW64 /c/Program Files/FreeCAD 1.0/bin
$ gdb freecad
GNU gdb (GDB) 16.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from freecad...
(No debugging symbols found in freecad)
(gdb) run
Starting program: C:\Program Files\FreeCAD 1.0\bin\freecad.exe
[New Thread 36064.0xad28]
[New Thread 36064.0xc2e4]
[New Thread 36064.0x7208]
warning: Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use:
   QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors.
   QT_SCREEN_SCALE_FACTORS to set per-screen DPI.
   QT_SCALE_FACTOR to set the application global scale factor.
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
warning: Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
[New Thread 36064.0xc208]
[New Thread 36064.0x65b0]
warning: onecore\windows\directx\database\helperlibrary\lib\directxdatabasehelper.cpp(855)\directxdatabasehelper.dll!00007FFC092D7223: (caller: 00007FFC092D682D) ReturnHr(1) tid(7098) 80004002 No such interface supported
warning: onecore\windows\directx\database\helperlibrary\lib\directxdatabasehelper.cpp(855)\directxdatabasehelper.dll!00007FFC092D7223: (caller: 00007FFC092D682D) ReturnHr(2) tid(7098) 80004002 No such interface supported
[New Thread 36064.0xb1e0]
[New Thread 36064.0xc2e8]
[New Thread 36064.0xa6d4]
[Thread 36064.0xb1e0 exited with code 1]
[New Thread 36064.0x93b8]
[New Thread 36064.0xbb10]
warning: onecore\windows\directx\database\helperlibrary\lib\directxdatabasehelper.cpp(855)\directxdatabasehelper.dll!00007FFC092D7223: (caller: 00007FFC092D682D) ReturnHr(3) tid(7098) 80004002 No such interface supported
[Thread 36064.0xbb10 exited with code 0]
[New Thread 36064.0xa504]
[New Thread 36064.0x82e8]
[New Thread 36064.0xc080]
warning: onecore\windows\directx\database\helperlibrary\lib\directxdatabasehelper.cpp(855)\directxdatabasehelper.dll!00007FFC092D7223: (caller: 00007FFC092D682D) ReturnHr(4) tid(7098) 80004002 No such interface supported
[Thread 36064.0xc080 exited with code 0]
[New Thread 36064.0x8278]
[New Thread 36064.0xc3d0]
[New Thread 36064.0xc210]
[New Thread 36064.0x9530]
[New Thread 36064.0xc330]
[New Thread 36064.0xbd84]
[New Thread 36064.0xbc7c]
[New Thread 36064.0x1638]
[New Thread 36064.0x938c]
[New Thread 36064.0xbde8]
[New Thread 36064.0xc2b8]
[New Thread 36064.0xbd10]
[New Thread 36064.0x5bf8]
[New Thread 36064.0xb0b4]
[New Thread 36064.0xa0a0]
[New Thread 36064.0xc0d4]
[New Thread 36064.0x5d4]
[New Thread 36064.0xc3bc]
[Thread 36064.0xbc7c exited with code 0]
[Thread 36064.0x938c exited with code 0]
[New Thread 36064.0x6448]
warning: onecore\windows\directx\database\helperlibrary\lib\directxdatabasehelper.cpp(855)\directxdatabasehelper.dll!00007FFC092D7223: (caller: 00007FFC092D682D) ReturnHr(5) tid(7098) 80004002 No such interface supported
[Thread 36064.0x6448 exited with code 0]
[Thread 36064.0x9530 exited with code 0]
[Thread 36064.0xc2b8 exited with code 0]
[Thread 36064.0xc330 exited with code 0]
[Thread 36064.0xbd84 exited with code 0]
[Thread 36064.0x1638 exited with code 0]
[Thread 36064.0xbde8 exited with code 0]
[Thread 36064.0x5bf8 exited with code 0]
[Thread 36064.0x93b8 exited with code 0]
[New Thread 36064.0xc3cc]

Thread 34 received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 36064.0xc3cc]
0x00007ffc0edddd11 in ntdll!DbgBreakPoint () from C:\WINDOWS\SYSTEM32\ntdll.dll
(gdb) bt
#0  0x00007ffc0edddd11 in ntdll!DbgBreakPoint () from C:\WINDOWS\SYSTEM32\ntdll.dll
#1  0x00007ffc0edefdce in ntdll!DbgUiRemoteBreakin () from C:\WINDOWS\SYSTEM32\ntdll.dll
#2  0x00007ffc0d3ae8d7 in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#3  0x00007ffc0ed714fc in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#4  0x0000000000000000 in ?? ()
(gdb)

The application is still running in gdb, haven't closed the gdb instance yet.

xtemp09, 2025-04-02:
This issue reminds #19023. @prashanthsmurthy, does your computer have multiscreen configuration? 

If so, does _FreeCAD_ work on the primary screen?

prashanthsmurthy, 2025-04-02:
Yes, I do have a dual monitor setup and now that you asked, I did try and application seemed to work fine too!  Thank you!  I honestly did not know which is primary and which is secondary monitor, but when I tried to open FreeCad in the other monitor, it worked fine.  I have just tried clicking parametric part and it moved to the Parts Design axis selection page.  I will let you know if I run into any other issues.

xtemp09, 2025-04-03:
@prashanthsmurthy, could you share details about the graphics card(s) you have, including the GPU driver version(s)?

prashanthsmurthy, 2025-04-03:
Hi, are there commands available to retrieve them in Win 11 Pro?  Here's what I found in System Information.  If you need anything else, do let me know how to retrieve it.  I will be more than glad to provide you.

General Information:
[image]

Display Information:
[image]

xtemp09, 2025-04-03:
Nvidia GT 1030. There was Nvidia GTX 1080 in the issue #19023.

The list of known OS-specific issues contains similar cases. [^1]

[^1]: https://forum.freecad.org/viewtopic.php?t=30573

FEA-eng, 2025-10-04:
@prashanthsmurthy Can you confirm if this issue is still relevant?