FreeCAD/FreeCAD #20516: FreeCAD does not open Python files from the command line

RobertPHeller, 2025-03-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Earlier versions of FreeCAD would open Python files from the command line, but not since FreeCAD 21.2.  It is awkward / inconvient to have to use the file menu to load the python file every time.  I am debugging Python code and need exit and restart FreeCAD, because of Python caching of loaded python modules. 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (xcb)
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-29:
If I use:

`./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage /home/username/Data/FreeCAD/Macro/MyScript.py`

the script is executed as I would expect, are you getting no warnings/errors in the Report View or in the Terminal at all?

RobertPHeller, 2025-03-29:
Not on my aarch64 machine using 
FreeCAD_1.0.0-conda-Linux-aarch64-py311.AppImage.  I copied 
~/Downloads/FreeCAD_1.0.0-conda-Linux-aarch64-py311.AppImage to 
/usr/local/bin/FreeCAD1.0.0 (/usr/local/bin is in my path).

My shell is /bin/tcsh, but that should not matter.  I am running Debian 12 on 
a Raspberry Pi 5.

Screen is just empty, when FreeCAD comes up.

Terminal output is:

marchhare% FreeCAD1.0.0 `pwd`/FoodBin.py 
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.

At Sat, 29 Mar 2025 10:44:57 -0700 FreeCAD/FreeCAD ***@***.***> wrote:

> 
> Syres916 left a comment (FreeCAD/FreeCAD#20516)
> 
> If I use:
> 
> `./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage /home/username/Data/FreeCAD/Macro/MyScript.py`
> 
> the script is executed as I would expect, are you getting no warnings/errors in the Report View or in the Terminal at all?
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

RobertPHeller, 2025-03-30:
I found a workaround by creating a symlink:

```
marchhare% dir -l *.FCMacro
lrwxrwxrwx 1 heller heller 10 Mar 30 12:43 FoodBin.FCMacro -> FoodBin.py

```

Then starting FreeCAD with FoodBin.FCMacro works as expected...

FEA-eng, 2025-10-04:
@RobertPHeller Can you confirm if this issue is still relevant?

RobertPHeller, 2025-10-04:
No, so long as I use the FCMacro.  It is mildly anoying that I can't use .py for Python code when I want to use the command line to run it.  (Using .py *used* to work, but I guess that has been deemed a "security problem".)

FEA-eng, 2026-08-17:
Seems to be by design, but still relevant.