FreeCAD/FreeCAD #20522: FEM: Stress linearization is wrong

giorgiugo, 2025-03-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Stress linearization (Asme VIII Div.2 5-A.4.1.2 or EN-13445) is a calculation on a tensor (all components in local coordinates referring to SCL: hoop, normal and tangential and not global because there's the need to take into account the right components and exclude some of them) and not on a vector (the code instead calculates formulas on an array (YAxis Data of the DataAlongLine that becomes sValues in the formulas)). An equivalent stress cannot be used as input, nor any other quantity taken individually beacuse the purpose is to calculate the tensor of membrane and bending and after that calculate the equivalent stresses of membrane, bending and membrane+bending in order to perfom the verifications.
So the inputs must be all the six stress components and x,y,z of the points of line (also to perform the coordinate transformation).
The outputs von Mises and Tresca of membrane, bending and membrane+bending and peak.
If i'm not misunderstanding the code I suggest freezing the tool until it is rewritten in order to prevent people from using it.
I suggest also to doublecheck the code and validate post-processing calculation tools (from several people) before insert it in stable versions of FreeCAD because FEM (and CFD) features are often not like those in CAD where "what i see is what i get".
Thank you all.

### Full version info

```shell
All versions since FEM stress linearization exists.
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-03-31:
I agree this tool needs fixing (apart from this issue, it shouldn't rely on exact stress field names) and should be used carefully (maybe we should add a warning in the documentation) but I wouldn't remove it.

Here's one usage example: https://fatigue.pro/2023/02/freecad-fem-results-in-fatlab/

But I also made a test model:

[image]

Here are the results from FreeCAD (with SXX component selected):

[image]

And from Abaqus (with all components computed by default):

[image]

giorgiugo, 2025-04-03:
@FEA-eng I'm glad you agree. Maybe in the rewritten tool is convenient to insert some flags in the GUI to select the global components wanted (as in many commercial software, with all preselected by default), so an user who wants use only some components can do that.
In additions also flags to exclude "on the fly"  local components to bending stress in order to be consistent whit ASME that doesn't want stress parallel to SCL and in-plane shear stress.
But i suggest (if you agree) to delete as soon as possible the possibility to use the current tool starting from von Mises (and Tresca) because of course it's theoretically very wrong and this is the thing that is misleading because an user thinks to do stress linearization (on all components with transformation and all the right calculations) and have as results von mises of membrane and von mises of membrane+bending but it's not. There are many posts on the internet of people who are approaching to do it.
Of course it's not possibile to delete von Mises from the PostFilterDataAlongLine beacuse it's useful have a von mises probe on a line.

FEA-eng, 2025-11-25:
> [@FEA-eng](https://github.com/FEA-eng) probably adding a warning to the documentation or even something within FEM to let the user know of this handicap ?

I added a note to the documentation. For now, I wouldn't touch the code, because it's quite likely that it will be fixed in the next release cycle. We may want to look for more volunteers, though - this tool requires careful rework and is highly desired by users because there are pretty much no open-source alternatives.