FreeCAD/FreeCAD #20532: ModuleNotFoundError: No module named 'PySide2' in Help.  Win10, FC1.1-40971

pixelwaster, 2025-03-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Did not see this in the build before 40504. (Didn't use 504). I also saw this with another addon toScketch, but there were some more issues and I removed it for now. I did try to search for a similar issue but pyside2 returned 110 results total.

```
Log: Init:      Initializing D:\FreeCAD_1.1dev-40971\Mod\Help\.\... failed
Log: ----------------------------------------------------------------------------------------------------
Log: Traceback (most recent call last):
  File "<string>", line 142, in RunInitGuiPy
  File "D:\FreeCAD_1.1dev-40971\Mod\Help\.\InitGui.py", line 24, in <module>
    import Help
  File "D:\FreeCAD_1.1dev-40971\Mod\Help\.\Help.py", line 58, in <module>
    from PySide2 import QtCore
ModuleNotFoundError: No module named 'PySide2'
Log: ----------------------------------------------------------------------------------------------------
Err: During initialization the error "No module named 'PySide2'" occurred in D:\FreeCAD_1.1dev- 40971\Mod\Help\.\InitGui.py
Err: Please look into the log file for further information
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/112.198/1
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * Defeaturing 1.2.3
  * Draft
  * DynamicData 2.73.0
  * Fem
  * freecad.gears 1.3.0
  * GrabBag
  * Help 1.0.3
  * IconThemes
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Material
  * Measure
  * Mesh
  * MeshPart
  * MeshRemodel 1.10.37
  * OpenSCAD
  * OpenTheme 2024.9.1
  * Part
  * PartDesign
  * Plot 2024.11.26
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.7.22
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-31:
Basically the Help [Addon](https://github.com/FreeCAD/FreeCAD-Help/blob/55758abd7467a76d2986987b9415169cfbeb1d1f/Help.py#L58) is being used as opposed to the [built-in](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Help/Help.py#L58) Help module. So either disable the Addon when using the new Qt6 based builds or just remove it completely.

pixelwaster, 2025-03-31:
So when I added a new workbench, Help got "updated" automatically. This used to happen to the BIM workbench. (brb)
I replace the folder with the one from the 7z download; the issue was solved. 

Now I wonder if BIM wasn't disabled from the first run, before any workbenches were added, that BIM would also be "updated" ?

Activated the BIM workbench, added a new workbench, and now both Help and BIM have been "updated. BIM icon is now the 4 colored squares and launching it fails with Arch error,

```
Err: No module named 'Arch'
Err: Traceback (most recent call last):
  File "D:\FreeCAD_1.1dev-40971\Mod\BIM\.\InitGui.py", line 78, in Initialize
    import Arch
```
Same as before:
```
Log: Init:      Initializing D:\FreeCAD_1.1dev-40971\Mod\Help\.\... failed
Log: ----------------------------------------------------------------------------------------------------
Log: Traceback (most recent call last):
  File "<string>", line 142, in RunInitGuiPy
  File "D:\FreeCAD_1.1dev-40971\Mod\Help\.\InitGui.py", line 24, in <module>
    import Help
  File "D:\FreeCAD_1.1dev-40971\Mod\Help\.\Help.py", line 58, in <module>
    from PySide2 import QtCore
ModuleNotFoundError: No module named 'PySide2'
Log: ----------------------------------------------------------------------------------------------------
Err: During initialization the error "No module named 'PySide2'" occurred in D:\FreeCAD_1.1dev-40971\Mod\Help\.\InitGui.py
Err: Please look into the log file for further information
```

Maybe remove BIM and HELP from the Add-on?

semhustej, 2025-04-01:
The old (pre-1.0) BIM Add-on should not be used with FreeCAD 1.0 and higher. It creates conflicts the new BIM workbench.

The discussion is here: https://github.com/FreeCAD/FreeCAD/issues/14072

pixelwaster, 2025-04-01:
I wasn't as clear as mud as they say. When going to the Addon Manager, the New BIM is replaced with the old BIM with no action on my part.  This happens if I add a new workbench/macro or not. I keep copies of new BIM and Help in a folder so I can replace them when they are replaced by Addon.

No idea on how to do a PR to request removal of those two items. They aren't really needed anymore....

Roy-043, 2025-04-02:
There is something seriously wrong with your installation. All built-in workbenches are listed as add-ons. That is the root cause of this issue.

closed 2025-04-02 by maxwxyz