FreeCAD/FreeCAD #20542: FEM: Pipeline filters are deleted when rerunning an analysis

FEA-eng, 2025-03-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, rerunning an analysis deletes all the filters added to the pipeline. Only the pipeline itself is kept and updated. This is quite annoying because one has to redefine e.g. the Warp filter after each run.

@marioalexis84, @ickby and @maxwxyz FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-03-31:
>Related to https://github.com/FreeCAD/FreeCAD/issues/13892?

No, it's a different issue.

ickby, 2025-04-26:
I looked into it, and it seems to be a setting side effect. There exists the following preference: Fem->General->Results->"Keep results on re-run". If unchecked, the pipeline is removed, and a new one is created, which is of course empty. If checked, the pipeline with all its filters stays in place, and the result is loaded into it. Both these behaviors seem counter intuitive for the user and should be changed. 

Would the following proposal make sense for you?

1. If "Keep on rerun" is True, the pipeline should stay as is, incl. all filters, but the data should be overridden to the new result
2. If "Keep on rerun" is False, the old pipeline stays as is, but a new empty one is created for the new data 

@marioalexis84 : I think you planned to work on the result handling. Are you on this? I want to prevent double work

ickby, 2025-04-26:
@marioalexis84 : it also seems that the new solver panel behaves differently than the old one. It does always keep the existing result pipeline, incl. its filters, and overrides the data. It seems that it ignores the "keep result on rerun" setting.

FEA-eng, 2025-04-26:
> I looked into it, and it seems to be a setting side effect. There exists the following preference: Fem->General->Results->"Keep results on re-run". If unchecked, the pipeline is removed, and a new [...]

It depends if "Keep results" is supposed to mean "Keep old results data" or "Keep old results objects". The simplest and probably most intuitive approach would be to:
1. Keep the old results objects and create a new one with the current results if this preference is checked 
2. Overwrite the data in the existing results objects (without deleting any of them) if it's unchecked (IMO this should be the default setting).

luzpaz, 2025-05-14:
@marioalexis84 👆

marioalexis84, 2025-05-16:
@ickby sorry for not responding; I missed this comment.
The new SolverCalculiX doesn't yet take these parameters into account, but I plan to add them (I'm a bit behind on that task).
Work on it as much as you want, and I'll then adapt SolverCalculiX to your work.
I agree with @FEA-eng that the current terminology is somewhat confusing.

commit c9ca852a (2025-08-20): FEM: Keep result pipeline and only reload data. Fixes #20542
closed 2025-08-20 by chennes