FreeCAD/FreeCAD #20546: STEP Export Does Not Convert Dimensions

cray54, 2025-03-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is often useful to send files using the native units for the design. 

When exporting STEP/STP files, there is an option to select "Units for export of STEP". The file gets flagged with the correct units, but the values are unchanged. 

For example, I have created a cube measuring 10mm x 10mm x 10mm. I have exported this into three different files: 

1. Units mm, Scheme AP214
2. Units m, Scheme AP214
3. Units in, Scheme AP214

When opening these files, the cube should still be 10mm on each edge. Importing Example 1 does result in a 10mm cube. However, importing Example 2 results in a 10m cube, and importing Example 3 results in a 10in cube.

[Cube 10x10x10mm.zip](https://github.com/user-attachments/files/19539123/Cube.10x10x10mm.zip)

I have not done comprehensive testing. I found this error by sending a STEP file for a manufacturing quote and their import showed a 50.8" dimension instead of a 2" dimension. 

Thank you, Chris

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.54
  * fasteners 0.5.37
  * Ship 2024.11.26
```

### Subproject(s) affected?

None

### Anything else?

So far, this is the only reference of this sort of a problem that I have found: 
https://forum.freecad.org/viewtopic.php?p=340919

This may be related to other issues, I have searched but am not adept at doing so in this platform yet. 

Thank you, Chris

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-04:
could you please test in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

cray54, 2025-04-04:
@maxwxyz at your request, I just tested this in the latest Windows build 40971. The behavior persists. 

Further, I tried exporting AP242 STEP and importing, and the behavior was the same. I also tested AP214 with "Use legacy export function" and the behavior was the same. This isn't comprehensive testing, but at least those functions do not work correctly. 

Thank you, Chris

maxwxyz, 2025-07-09:
@pieterhijma is this an OCC issue for STEP export units?

ipatch, 2025-12-31:
i am unable to reproduce, however i am building freecad with occ v7.9.2

[image]

```
OS: Fedora Linux Asahi Remix 40 (Forty) (tty)
Architecture: arm64
Version: 1.2.0dev.44840 (Git)
Build date: 2025/12/30 17:40:39
Build type: Release
Branch: main
Hash: f5759e580b6330906beca12b4be419b030a088ab
Python 3.13.10, Qt 6.9.3, Coin 4.0.6, Vtk 9.5.2, boost 1_89, Eigen3 5.0.1, PySide 6.9.3
shiboken 6.9.3, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.2
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.958/1
```