FreeCAD/FreeCAD #20560: Draft Workbench Rectangular Array tool does not work when Unit System is Building US

proPonent, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Draft Workbench Rectangular Array tool does not work when Unit System is Building US.
Other Unit systems will create arrays in the draft workbench without issue, but Building US will fail every time.

### Full version info

```shell
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Fcmcua 1.0.0
  * freecad.gears 1.3.0
  * boltsfc 2022.11.5
  * LCInterlocking
  * MakerWorkbench 1.0.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * Curves.backup1739394411.6016512 0.6.57 (Disabled)
  * Curves 0.6.58
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-04-01:
I can confirm the issue. This is the error message:
```

17:44:39  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\drafttaskpanels\task_orthoarray.py", line 176, in accept
    self.v_z) = self.get_intervals()
                ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\drafttaskpanels\task_orthoarray.py", line 280, in get_intervals
    v_x = App.Vector(U.Quantity(v_x_x_str).Value,
                     ^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: syntax error
17:44:51  <class 'AttributeError'>
```

https://github.com/user-attachments/assets/f5f57ab9-0dff-49b8-ae73-275f9ae3eff6

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

Roy-043, 2025-04-05:
Related to: #11345.

closed 2025-10-23 by Roy-043