FreeCAD/FreeCAD #20561: TechDraw: Center lines do not display correct color

jlovett11211, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting a color for center lines in tech draw, the color is not displayed. No matter what color is selected the color remains visibily black. When I double click on the line, the color selection remains. This issue is new with the Windows 40971 version. 

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.047/1.5
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Gridfinity 0.5.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-04-01:
I'll take a look

benj5378, 2025-04-02:
Also affects non-centerlines

commit 2aba9a86 (2025-04-07): TechDraw: fix wrong color on edges

Fixes #20561
closed 2025-04-07 by hyarion