FreeCAD/FreeCAD #20564: Can't Import *.JPG File.

CFK520, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Part Design WB -> New Document -> New Body -> File -> Import -> *.JPG file
Then nothing will happen.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Chinese/Taiwan (zh_TW)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.1167/1.5
Installed mods: 
  * A2plus 0.4.68
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * Design456 0.0.1
  * ExplodedAssembly
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * woodworking 0.21.2.33771.0.24.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-04-02:
I think it would much better if the JPG file were uploaded.

CFK520, 2025-04-02:
> I think it would much better if the JPG file were uploaded.

Thanks and uploaded.
It seems works well on some other *.JPG files.
But this one not.
* Tried this on V1.0 without problem.

xtemp09, 2025-04-02:
No problem here. Note that I have the same _Windows_ version and the same weekly bundle.

<details>

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41029 (Git)
Build date: 2025/03/28 15:39:34
Build type: Debug
Branch: master
Hash: db8b58f73eb1935718efce3b5046017881853d64
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96.0219/95.9748/1
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41078 (Git)
Build date: 2025/03/30 16:20:33
Build type: Debug
Branch: iss-20498-BIM
Hash: 598168f683eab642213b88363b17a98365a78553
Python 3.12.3, Qt 6.7.2, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9748/1
```

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.706/1
```

</details>

luzpaz, 2025-09-26:
Attachment was never legitmately attached. It needs to be dragged into this ticket in order to attach.  
Closing as there was no response from OP. If a jpg is provided we'll re-open ticket.

closed 2025-09-26 by luzpaz