FreeCAD/FreeCAD #20569: Add adjustable line width for constraint guides

Beep6581, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot find a way to adjust the line width of constraint line - those red lines with and without arrows:

[image]

This leads to a messy preview.

Please add an option to allow adjusting the line width and color of those lines.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda AppImage
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * fasteners 0.5.37
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-04-04:
@FreeCAD/design-working-group This is required for the new sketcher style.

Beep6581, 2025-04-08:
Is there an issue I can keep track of?