FreeCAD/FreeCAD #20580: Removing splitter failed

Beep6581, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Create simple centered square in Part Design, 20x20mm.
2. Pad it 20mm.
3. Create centered circle on a face, 3mm.
4. Create hole, depth 15mm, standard ISO Metric Fine, size M12x1.5, threaded, model thread, ok.
5. "Removing splitter failed".

If you set depth to 20mm, it works.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41080 (Git)
Build date: 2025/03/31 13:17:51
Build type: Release
Branch: main
Hash: ee6f757c577b20b5bfe29ebee26a0651a7c7c921
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.38
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.13
  * Curves 0.6.60
```

### Subproject(s) affected?

None

### Anything else?

[threadpitch.zip](https://github.com/user-attachments/files/19555999/threadpitch.zip)

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-01:
The `Removing splitter failed` error come from the refinement, try to switch the `Refine` property. If it's raised at object creation then you have to switch the auto-refinement preference.
Related #18110 #17846

maxwxyz, 2025-04-04:
Im closing this in favor of the mentioned existing issues.