FreeCAD/FreeCAD #20585: Interface: Tasks bar text color too dark

ErtugrulEge, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,
Tasks bar has default  black text color with dark blue background so reading is difficult.The default text color should be white or some light colors

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-02:
> Stylesheet/Theme/**QtStyle**: unset/FreeCAD Classic/**windowsvista**

Did you tinker with the Qt settings ?

ErtugrulEge, 2025-04-02:
Version 1.0 has no problem, also if I change to FC light or Dark theme on latest weekly I don't have any problem it is just about classic theme 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

furgo16, 2025-04-02:
I can confirm on Linux as well, with latest build and Classic theme

maxwxyz, 2025-04-02:
@MisterMakerNL FYI

MisterMakerNL, 2025-04-02:
Couple weird stuff...
First are we building in QT 6 for now?
Why does Windows 10 has Vista Qtstyle.
It isn't used for years.

furgo16, 2025-04-02:
While the OP has Qt 6, I get the same issue with Qt 5.

[image]

Could perhaps one of these PRs have something to do with it? https://github.com/FreeCAD/FreeCAD/pull/19649 https://github.com/FreeCAD/FreeCAD/pull/19292? @alfrix, would you know? 

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41130 (Git) Snap 1361
Build date: 2025/04/02 04:19:53
Build type: Release
Branch: main
Hash: f4adf2e0e937989bbea2bf64659b2fd622dfd3d7
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

alfrix, 2025-04-02:
the text in the header should be using highlighted text color, which in this case would be white, but it seems to not be working for some reason, maybe the palette here makes selected text black instead of the traditional inverted

the solution is simple, remove the accent color from the header and change it something else since it's not possible to manage it properly without qt6.5 (the notification about system palette changes is only implemented after that version), and since 1.1 needs to support qt6.2

i already have almost done some fixes that should solve this, i only need to tweak it a little more before making a PR.

MisterMakerNL, 2025-04-21:
@alfrix You made the highlight color darker here: https://github.com/MisterMakerNL/FreeCAD_MisterMaker/blob/31b0dbaa79da9e0a8d18c009006a2b1c6cd16404/src/Gui/QSint/actionpanel/actionpanelscheme.cpp#L45
I think if you do not do this it might just work out.
I also noticed that the style settings of the border remove the border completely and even for theme's. 
It is kinda weird that we now have a fold-up button/header without the part that folds up visible.
We could decide to remove the fold-up mechanisme I am not using it. But maybe someone else?

MisterMakerNL, 2025-04-21:
But there is a text-bakcground color and a background-color that is used for toolbars and empty panel space.
You shouldn't put text on the panel background.
Should ether promote the complete panel background to text background color or add a background to the text.

MisterMakerNL, 2025-04-21:
That is hell of a job over Linux,OSX and Windows.
I know Realthunder uses in the overlay system a simple check if the stylesheet has "Dark" in the name and then set the appropriate stylesheet for overlay.
Although that doesn't really work since it is tied to the background color of the 3D view, it could work tho.
Also QT has some tips here: https://www.qt.io/blog/dark-mode-on-windows-11-with-qt-6.5

PR #21049 (2025-05-21): Gui: fix light/dark switch when on classic
This fixes the taskpanel issues to transition from light to dark when the system changes

fixes the double icon mentioned in #21036
fixes #20585
should fix the taskpanel part of #15346

[Videocaptura de pantalla_20250511_162917.webm](https://github.com/user-attachments/assets/c3177404-9066-40a8-8450-b76a4d28f8b1)
closed 2025-05-21 by kadet1090